robotology/community #140: using `cartesianInterfaceControl`

matejhof, 2016-09-13:
With iKinCartesianSolvers running, following the info [here](http://wiki.icub.org/brain/group__cartesianInterfaceControl.html), I'm doing the following: 
```cartesianInterfaceControl --part left_arm --onlyXYZ```
and from another terminal:
```  yarp write ... /armCtrl/left_arm/xd:i```
```> (-0.18 -0.1 0.1 1.0 0.0 0.0 0.0) ```
(I tried with and without brackets)
But there is no response.
Can someone, most likely @pattacini :), please tell me what is wrong?

pattacini, 2016-09-13:
@matejhof just tested on my system against `icubSim` and it's perfectly working.
Double check that all the required resources are running fine on the robot / your system.

If you launch the module with `--onlyXYZ`, then you'd need only 3 doubles (without brackets).

pattacini, 2016-09-14:
Should work also with more than 3 doubles in `--onlyXYZ`; see https://github.com/robotology/icub-main/blob/master/src/modules/cartesianInterfaceControl/main.cpp#L276.

closed 2020-12-15 by pattacini