robotology/community #142: Cartesian controller server configuration for Vizzy

plinioMoreno, 2016-09-22:
Vizzy's [cartesian controller server code](https://github.com/vislab-tecnico-lisboa/vizzy/blob/master/vizzy_yarp_icub/src/modules/vizzyCartesianControllerServer/main.cpp), is configured with [this file](https://github.com/vislab-tecnico-lisboa/vizzy/blob/master/vizzy_yarp_icub/app/vizzyCartesianControllerServer/server_left_arm.ini).

I want to know if the code is still valid (it is pretty old in fact), or it needs to be updated to the recently added control interfaces (e.g. position direct, etc) and the most recent version of the firmware. Vizzy's arms can be controlled only in position/position direct/velocity because it does not have torque sensors.

pattacini, 2016-09-25:
Hi @plinioMoreno 

Actually, the **yarp code** you posted is fine. It's already using `position-control` by default and you may want to highlight that introducing this [_new parameter_](https://github.com/robotology/robots-configuration/blob/master/iCubGenova01/cartesian/left_arm_cartesian.xml#L10) in the conf file, also in order to disable it just on purpose.

Concerning **firmware**, I don't know if `Vizzy` makes already use of the latest [Control and Interaction Modes](http://wiki.icub.org/wiki/Control_Modes) policies.

Finally, the `Cartesian Control` doesn't deal with `Torque Control` since the latter is handled at a lower level.

plinioMoreno, 2016-09-30:
Hi @pattacini,

Thanks for your answer. Vizzy actually can control the joints in the mixed and the position direct modes. I will set the PositionControl flag and test when the [segfault issue](https://github.com/robotology/yarp/issues/914) is fixed.

I see that the cartesian controller server and solver run with yarprobotinterface for iCub. Would it be possible to run the cartesian controller and server for Vizzy from the yarprobotinterface? If yes, how should I do it?

pattacini, 2016-10-02:
Hi @plinioMoreno, 

You can definitely launch the Cartesian servers devices using the **`yarprobotinterface`** deployer (see also #146). You would need to populate **Vizzy**'s configuration files with suitable `xml` containing corresponding options. To get an idea, look at the content of this [directory](https://github.com/robotology/icub-main/tree/master/app/simCartesianControl/conf).

By contrast, the Cartesian solvers still need to be run as separate binaries, given that they are not devices.

pattacini, 2016-10-05:
Hi @plinioMoreno 

Can we close this?

plinioMoreno, 2016-10-06:
Yes, we can close it.

closed 2020-12-15 by pattacini