robotology/community #147: Torso oscillates slightly after initialization

paraschos, 2016-10-13:
We recently updated the firmware of iCubDarmstadt01 and now after the normal boot the torso oscillates a bit. What could the issue be? Gains? However, the gains in the config files didn't change, the firmware did. Also icub-main but I am not sure if it affects the robot since the eth boards do the ctl.

- The current configuration (with the changes to the previous one) is shown here

https://github.com/robotology/robots-configuration/pull/6

- The last icub-main we used with stable behavior is 

https://github.com/robotology/icub-main/commit/4a8c4f2e34980153d98b2d06a31ada8d33736e01

Any ideas? Shall I just play a bit with the gains?

pattacini, 2016-10-13:
Hi @paraschos 

I wouldn't focus on `icub-main` indeed, as you said, but rather on `FW` and robots configuration.

First off, I'd try to **get to a condition where the oscillation is no longer there**, by (1) _reverting the conf files_ and then (2) _rolling back the previous `FW`_ (I don't know how much this would be doable/cumbersome). At least these two steps should likely help narrow down the problem.

Tuning PID gains should be the last resort.

cc @robotology/icub-support-team 
@davidetome could you please double check these changes https://github.com/robotology/robots-configuration/pull/6?

paraschos, 2016-10-13:
Hi @pattacini, 

I am afraid that such a process will take many many hours: in order to make the robot boot I would
have to re-flash quite a few boards as the latest `icub-main` requires the 2foc with protocol 1.5. Now if I revert to a previous firmware on the torso, I would have to revert `icub-main`. But the reverted `icub-main` will not now the v1.5 protocol of the rest of the joints and will seg fault. That means reflashing  the rest of the robot with, at least, the 2foc old driver. Not sure if I also get problems with skin, strain and/or mais.

A good idea would be to have some automatic way of reflashing the whole robot but it is not the case.

Update on the problem: the oscillations occur only near the zero position of the torso. All other configurations that I tried were stable. Only the pitch (3rd joint) seams to oscillate. 

I checked the gains a bit, how did you ended up with values like Kp=1066.66? The appear quite precise to be hand set. Why there is no Kd part? My tune was to increase the derivative gain to Kd=150 and it performs quite better, but not fully cured.

pattacini, 2016-10-13:
Ok @paraschos understood. I proposed to do the roll back just to be 100% sure that nothing happened to the mechanics and/or electronics (e.g. encoder) in the meantime.

I suggested to tune PID only as last resort because those values should usually cover a wide range of conditions on different robots of the same configurations.

I honestly ignore how `P`, `I`, and `D` are retrieved; I don't know if a proper automatic procedure is used. Others (e.g. @randaz81) can chime in here.

We have some code to perform automatic tuning (see #2), though I believe it's not employed in this case.

The `D` gain is not always necessary as `P` and `I` are: it depends on a case basis. The oscillations can be definitely damped off by means of the derivative; my only concern was only to rule out the possibility that they are not caused by the interaction of new `FW` with old gains (you said they didn't change).

paraschos, 2016-10-14:
@davidetome Any updates, are the config files good? 
Also some questions what is the "build" option that we added for? What was wrong 
left/right MAIS device setting causing?

davidetome, 2016-10-17:
Hi @paraschos , 

 the configuration files seem to be good.

pattacini, 2016-10-17:
The `build` param is a placeholder used for version checking, whereas the `MAIS` are those boards controlling the acquisition of the fingers encoders. Missing or wrong values for these specific options could bring about _warnings_ and/or _errors_ while launching `yarprobotinterface` so as loading incorrect parameters for one of the hands.

However, I don't think they can interfere with the problem under inspection.

paraschos, 2016-10-24:
@pattacini I want to take one more look today on this one. Then, if there is no resolution I am closing it.

marcoaccame, 2016-10-24:
Hi @pattacini and @paraschos,
if it can help, last week I tested on our robot iCubGenova02 the master release (we normally run the devel release on this robot) and I did not see any oscillation on torso. 

I was aware of this issue, hence I looked at (and moved) the torso trying to find wrong movements. But I saw nothing strange.

pattacini, 2016-10-24:
Thanks @marcoaccame for testing.

paraschos, 2016-10-24:
@marcoaccame ty!

paraschos, 2016-10-24:
Update: the problem appears to be more general than the torso oscillation. We did the following:

- Disable all the motors (idle) but the right leg. Small oscillations were still present
- The same for the left leg
- ... and for the torso

We noticed that the oscillations are mainly due to the `pitch` joints. Playing around with the 
leg PID controllers (position mode) we noticed that disabling the `K_i` did the trick. I am not 
sure if there is any other scaling of the gains, but `K_i~10K` with clipping to `1.5K` can lead
to bang-bang ctl with the clipped value.  We send some test trajectories to the legs j:0 and
tracking/reaching was relatively OK. What is the reasoning behind these high `K_i` values?

pattacini, 2016-10-25:
Hi @paraschos 

What was the purpose of idling other joints when testing one specific?

Further, we could argue that `K_i` is too high but we cannot shut it down outright, given that we really need the integral term to compensate for steady-state errors due impairments, gravity...

The output of the `I` part is not clipped, otherwise we have overshoots and oscillations. The integral gets disabled and does no longer accumulate against the input error as long as we are beyond a specific threshold (this is the `anti-windup` mechanism) .

That being said, for knowing why we have those gains we would need to ask those who have been working with legs and/or tuning PID @randaz81 @traversaro @DanielePucci @francesco-romano. Thy might have more insights.

paraschos, 2016-10-26:
Hi @pattacini 

> What was the purpose of idling other joints when testing one specific?

To figure out which joint/joints are the cause of these subtle oscillations.  Additionally, by decoupling the system we avoided 'cross-talking' between the joints.

> Further, we could argue that K_i is too high but we cannot shut it down outright, given that we really need the integral term to compensate for steady-state errors due to impairments, gravity...

I only did set it to zero for testing. What I also wanted to state is that the P gains is quite high and, therefore, the robot goes 'close-enough' (well that is a different discussion) to the desired. 

> The output of the I part is not clipped, otherwise we have overshoots and oscillations. The integral gets disabled and does no longer accumulate against the input error as long as we are beyond a specific threshold (this is the anti-windup mechanism) .

I am not sure if I fully understand this part. So there is no clipping in `K_i`, but the `\dot{K_i}` gets on/off 
when the ` | q_real - q_des | <= threshold `? What is the `Integral Limit` affecting?

Bottom line, I do not see any problem with the robot, maybe a bit of gain tuning to be done. When I find some time I will play around to find suitable gains for our setup (and since we mainly use torque control but for simple movements, we don't have high requirements for accurate tracking) and give you an update.

pattacini, 2016-10-26:
A pure `P` controller cannot reject external disturbances, neither compensate for uncertainties and unmodeled quantities, so the **close enough** you saw in some given experiments is definitely and unfortunately **not enough** for general usage.

I think you'll find this [resource](https://it.mathworks.com/help/simulink/examples/anti-windup-control-using-a-pid-controller.html) on the **anti-windup** mechanism, which comes accompanied with helpful diagrams and models, quite enlightening.

In short, the anti-windup serves to correctly deal with actuators' saturation (`max output` parameter) and even to further constrain the integral output (`integral limit` parameter < `max output`). In fact, in presence of such non-linearity, we must somehow notify the `I` block that its output won't be fully applied to the motor because of the saturation.

Basically, the anti-windup boils down being a feedback line connecting the `I` output to the `I` input through some logic entailing as net result to stop the integration while we are in the regime of saturation.

giuliavezzani, 2016-12-16:
@julijenv  and I experienced oscillations when the torso joint no. 2 is at 0.0 degree.
If we move the torso to a different position, for example, at some negative values, the torso became stable. However. while using the torso, such a instability comes out again, maybe because of crossing of the 0.0 degree value.

closed 2020-12-08 by pattacini