robotology/community #148: Travis not loading custom yarp plugins

jgvictores, 2016-10-13:
Hi! We've finally been moving on to shared libraries for YARP plugins within the [teo-main](https://github.com/roboticslab-uc3m/teo-main) repository. Tests are running fine on my local machine, but when I push them to Travis, devices cannot be loaded (tests crash in segfaults). The following below is Travis output of the `issue-60` repository branch, where I've been adding debug code to [.travis.yml](https://github.com/roboticslab-uc3m/teo-main/blob/4983d72004da64569735164244317513095ff55f/.travis.yml) to try to discover what's happening. Everything before the tests seems well... Any ideas on how to further debug (or fix ^^) this?

([source of first code that breaks after determining device not valid on Travis](https://github.com/roboticslab-uc3m/teo-main/blob/4983d72004da64569735164244317513095ff55f/test/testKdlSolver.cpp)) <- used to work with static libraries and `YARP_DECLARE_PLUGINS`+`YARP_REGISTER_PLUGINS`.

```bash
The command "sudo make install" exited with 0.

$ sudo ldconfig
The command "sudo ldconfig" exited with 0.

$ yarp plugin --list
BasicCartesianControl
CartesianControlClient
CartesianControlServer
FakeControlboard
FakeControlboardOR
KdlSolver
The command "yarp plugin --list" exited with 0.

$ yarp plugin --verbose KdlSolver
Yes, this is a YARP plugin
Trying to find library '/usr/local/lib/yarp/KdlSolver.so' containing function 'KdlSolver' -- found
  * library:        /usr/local/lib/yarp/KdlSolver.so
  * system version: 5
  * class name:     teo::KdlSolver
  * base class:     yarp::dev::DeviceDriver
The command "yarp plugin --verbose KdlSolver" exited with 0.

$ ldd /usr/local/lib/yarp/KdlSolver.so
	linux-vdso.so.1 =>  (0x00007ffe9eec5000)
	libYARP_OS.so.1 => /usr/local/lib/libYARP_OS.so.1 (0x00007f3da4ac2000)
	liborocos-kdl.so.1.3 => /usr/local/lib/liborocos-kdl.so.1.3 (0x00007f3da4837000)
	libKdlVectorConverterLib.so => /usr/local/lib/libKdlVectorConverterLib.so (0x00007f3da462a000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f3da432a000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f3da4114000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3da3d56000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3da3b39000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f3da3931000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f3da372d000)
	libreadline.so.6 => /lib/x86_64-linux-gnu/libreadline.so.6 (0x00007f3da34eb000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3da31ef000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f3da50db000)
	libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007f3da2fc8000)
The command "ldd /usr/local/lib/yarp/KdlSolver.so" exited with 0.

$ bin/testKdlSolver
Running main() from gtest_main.cc
[==========] Running 8 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 8 tests from KdlSolverTest
[ RUN      ] KdlSolverTest.KdlSolverFwdKin1
yarpdev: ***ERROR*** could not find device <KdlSolver>
[error] testKdlSolver.cpp:26 SetUp(): solverDevice not valid: KdlSolver.
/home/travis/build.sh: line 45: 21070 Segmentation fault      (core dumped) bin/testKdlSolver
The command "bin/testKdlSolver" exited with 139.
```

pattacini, 2016-10-14:
Maybe @drdanz can help here?

drdanz, 2016-10-14:
That's weird... Try adding `YARP_VERBOSE=1` to the latest command, it should give some more information about the reason why the plugin cannot be found...

drdanz, 2016-10-14:
i.e. run it as `YARP_VERBOSE=1 bin/testKdlSolver`

jgvictores, 2016-10-17:
So... the issue has to be somewhere between YARP, travis, and gtest...

A verbose YARP would definitely help.

However, the following 4 approaches have not made my tests any more verbose regarding YARP:
- .travis.yml `- YARP_VERBOSE=1 bin/testKdlSolver`
- .travis.yml `env:` `global:` `- YARP_VERBOSE=1` ([here in the TRAVIS output](https://travis-ci.org/roboticslab-uc3m/teo-main/jobs/167729705) it can be seen that it even affects `yarp plugin --list`, but still not `bin/testKdlSolver`)
- testKdlSolver.cpp `yarp::os::Property solverOptions("(verbose 1)");`
- testKdlSolver.cpp `setenv("YARP_VERBOSE","1",1);`

I'd like to call/force something like `yarp::os::YarpPluginSettings::setVerboseMode(true)` but it's way out of scope and not static :sweat_smile:... any other ideas/alternatives?

drdanz, 2016-10-18:
you can set the `ResouceFinder` to be verbose by calling explicitly [`ResourceFinder::setVerbose()`](http://www.yarp.it/classyarp_1_1os_1_1ResourceFinder.html#af6b29f542e049ca102f1d871cee3a627)

jgvictores, 2016-10-20:
I've made a simple [example repository](https://github.com/jgvictores/example-yarp-plugin-gtest) with a simple [example program](https://github.com/jgvictores/example-yarp-plugin-gtest/tree/master/programs/exampleProgram) that attempts to load a simple [example plugin](https://github.com/jgvictores/example-yarp-plugin-gtest/tree/master/libraries/ExampleLibrary) (a YARP plugin compiled as a shared library).
- **On my machine**: The example program opens the example plugin. The `YARP_VERBOSE` activates some "Scanning. I'm scanning." console output in `yarp plugin`, *and* on the example program.
- **On Travis (gcc/clang)**: The example program **cannot** open the example plugin (`yarpdev: ***ERROR*** could not find device <ExampleLibrary>` can be seen on the [travis job log](https://travis-ci.org/jgvictores/example-yarp-plugin-gtest/jobs/168855396)). The `ResourceFinder` verbosity has no effect on the plugin mechanism in general. The `YARP_VERBOSE` activates some "Scanning. I'm scanning." console output in `yarp plugin`, but strangely enough it *does not activate this console output* in the example program on Travis (even though `std::getenv("YARP_VERBOSE")` says it is set at runtime).

I'm thinking in multi-threading, just because illegal instructions should be caught in exit return codes... Could you think of any architecture-dependent feature in the guts of yarp plugins that may not be available on Travis?

PeterBowman, 2016-11-03:
@jgvictores did you compile YARP with `SKIP_ACE` option set to `ON` on your local machine? (see [.travis.yml](https://github.com/jgvictores/example-yarp-plugin-gtest/blob/b4b45f2/.travis.yml#L25)) In my case it doesn't even try to load the driver because of [this](https://github.com/robotology/yarp/blob/d790dbc/src/libYARP_dev/src/Drivers.cpp#L295). Everything works fine with ACE enabled.

drdanz, 2016-11-08:
FYI This issue with `SKIP_ACE` is fixed on devel by robotology/yarp@d117fffc0392a5cb13a7b0b9352b5825a0aa4785

jgvictores, 2016-11-08:
@drdanz Ok, thanks for the pointer!
@drdanz @PeterBowman Thanks for the help to both of you!

closed 2020-12-15 by pattacini