robotology/community #151: New GUI ethLoader

hu-yue, 2016-10-21:
Hi, 
I wanted to update the robot, and as I've understood from issues #134 and #135 that there is a new gui for the ethLoader, and all the updated instructions are referred to the nre gui.
However even if I updated everything I still see the old gui, is there any option or so that I had to active to see the new gui?
(I updated the firmware anyways with the old one, but for future and also students it would be better to have the one that matches with the instructions)

Thanks!

marcoaccame, 2016-10-24:
Hi @hu-yue, in master there is not really a new GUI for ethLoader. 

Some months ago, I just did a minor change to the layout: I reorganized the column to show some more information and I have renamed some buttons. 

Following this change, new instructions were also required. The instructions refer to the new layout because one should typically perform an update of everything, as described in https://github.com/robotology/QA/issues/135.

So, the ethLoader GUI is not new, it just shows some more information. However, to have the new information displayed one also need to update the FW of the eLoader and eUpdater to versions equal or higher than 2.20. 

Should one update also the FW of eLoader and eUpdater? I did not mention it because it is not necessary yet. See issue https://github.com/robotology/QA/issues/134 for reasons.

Here are some screenshot so that you can better understand the changes of layout and what happens if you update eLoader / eUpdater or not. 

[image]

**Figure 1** - Previous GUI for ethLoader.

[image]

**Figure 2** - Current GUI for ethLoader.  It has extra columns about type of board, running process, date of FW and user-defined info of the board. Board 10.0.1.1 does not show the new information because it has old versions of eLoader and eUpdater, whereas the other boards do because they have  eLoader / eUpdater with version  >= 2.20.

hu-yue, 2016-10-24:
@marcoaccame thanks for the clarification!
We will try to update the eLoader and eUpdater later on, I will close this issue for the time being.

marcoaccame, 2016-10-24:
Hi @hu-yue ,

when you update eLoader, **remember** that instructions tells: 
`In doing this operation, be careful and update only one board at a time because with some upgrades the  board may change its address into 10.0.1.99`.
Well, this update does force a reset of the address, hence ... do it one board at a time otherwise all boards will have the same IP address.
Also, if you change the IP address, you need to restart the board to see the new address in the GUI.

closed 2020-12-15 by pattacini