robotology/community #152: How to use the iCubFinger forward kinematics

pattacini, 2016-10-27:
The [**`iCub::iKin::iCubFinger`**](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubFinger.html) object is designed to compute the forward kinematics of the iCub fingers.

Since there is no direct mapping among the hand encoders streamed by the standard state port, e.g. **`/icub/left_arm/state:o`**, and the joints of the chain representing the specific finger, we can make use of [`iCub::iKin::iCubFinger::getChainJoints()`](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubFinger.html#ae43422b983bd35f02b64732ceb3d50a8) helper method as done here below:

```cpp
yarp::os::Property options;
options.put("device","remote_controlboard");
options.put("remote","/icub/left_arm");
options.put("local","/local/left_arm");

yarp::dev::PolyDriver driver;
iCub::iKin::iCubFinger finger("left_index");

if (driver.open(options))
{
    yarp::dev::IEncoders *iencs;
    if (driver.view(encs))
    {
        int nEncs;
        iencs->getAxes(&nEncs);
        yarp::sig::Vector encs(nEncs);
        iencs->getEncoders(encs.data());

        yarp::sig::Vector joints;
        finger.getChainJoints(encs,joints);
        Matrix tipFrame=finger.getH((M_PI/180.0)*joints);
    }
}
```

The helper function simply takes the encoders as input, **extracts relevant components** referring to the correct finger (the _index_ in this case), and applies **suitable transformations** to them.

The resulting joints can then be used to populate the kinematic chain straightaway. Notice the **conversion from degrees to radians**.

pattacini, 2016-10-27:
The **`/icub/left_arm/state:o`** streams out only one double accounting for the movements of the two underactuated distal joints of the fingers. However, relying on the measurements acquired from the `MAIS` boards, we can have access to the two "_hidden_" encoders of each distal phalanxes.

These further feedback can be profitably used to improve the accuracy of the forward kinematics computation. We can thus specialize the snippet above for the _left index_ as in the following:

```cpp
yarp::os::Property options2;
options2.put("device","analogsensorclient");
options2.put("remote","/icub/left_hand/analog:o");
options2.put("local","/local/left_hand");

yarp::dev::PolyDriver driver2;
if (driver2.open(options2))
{
    yarp::dev::IAnalogSensor *ianalog;
    if (driver2.view(ianalog))
    {
        yarp::sig::Vector analogs;
        ianalog->read(analogs);

        // image you have here the Vector joints
        // as prepared from the snippet above
        // with resort to getChainJoints() helper

        // apply corrections
        joints[2]=analogs[4];
        joints[3]=analogs[5];

        Matrix tipFrame=finger.getH((M_PI/180.0)*joints);
    }
}
```

The distal joints of the left index (the last two components) are corrected resorting to the analog readings.

The [**`http://www.yarp.it/classyarp_1_1dev_1_1IAnalogSensor.html`**] interface provides a Vector of 15 double per hand in this case, which are arranged according the these rules:
- **3 double per finger** accounting in order for the **proximal** and the **two distal** phalanxes.
- the fingers' order is: **thumb**, **index**, **middle**, **ring**, **pinkie**.

pattacini, 2016-10-27:
cc @vtikha @claudiofantacci @giuliavezzani @nawidj

pattacini, 2016-10-28:
THIS IS AN EDIT OF THE PREVIOUS MESSAGE

The **`/icub/left_arm/state:o`** port streams out only one double accounting for the movements of the two underactuated distal joints of the fingers. Moreover, the angles provided by this port do refer to **motors encoders**.

However, relying on the measurements acquired from the `MAIS` boards, we can have access to the two "_hidden_" encoders of each distal phalanxes.

The `MAIS` feedback can be profitably used to **improve the accuracy of the forward kinematics computation** also because they account for **joints encoders** that do sense what motor encoders cannot (e.g. tendon flexibility). We can thus specialize the snippet above for e.g. the _left index_ as in the following:

```cpp
yarp::os::Property options2;
options2.put("device","analogsensorclient");
options2.put("remote","/icub/left_hand/analog:o");
options2.put("local","/local/left_hand");

yarp::dev::PolyDriver driver2;
if (driver2.open(options2))
{
    yarp::dev::IAnalogSensor *ianalog;
    if (driver2.view(ianalog))
    {
        yarp::sig::Vector analogs;
        ianalog->read(analogs);    // analogs are expressed in 0...255 range => we need conversion

        // imagine you have available here the Vector joints
        // as prepared from the snippet above with resort to
        // iCub::iKin::getChainJoints() helper

        // get access to joints bounds
        iCub::iKin::iKinChain &chain=*finger.asChain();
        
        // apply corrections
        // joints[0] is the abduction/adduction for which we don't have any MAIS readout
        joints[1]=(analogs[3]*(chain[1].getMax()-chain[1].getMin())+chain[1].getMin())*(180.0/M_PI);
        joints[2]=(analogs[4]*(chain[2].getMax()-chain[2].getMin())+chain[2].getMin())*(180.0/M_PI);
        joints[3]=(analogs[5]*(chain[3].getMax()-chain[3].getMin())+chain[3].getMin())*(180.0/M_PI);

        Matrix tipFrame=finger.getH((M_PI/180.0)*joints);
    }
}
```

The proximal and distal joints of the left index (the last 3 components) are corrected resorting to the analog readings, whereas we have to still use the motor encoder for the abduction/adduction (first component).

The [**`IAnalogSensor`**](http://www.yarp.it/classyarp_1_1dev_1_1IAnalogSensor.html) interface wraps around the **`/icub/left_hand/analog:o`** port, providing a Vector of 15 numbers[¹](http://wiki.icub.org/wiki/Software_interface:_standard_port_names_for_hardware_devices#Finger_encoders) in the range `0...255` that are arranged according to these two rules:

1. **3 numbers per finger** accounting in order for the **proximal** and the **two distal** phalanxes.
2. the fingers' order is: **thumb**, **index**, **middle**, **ring**, **pinkie**.

#### References
1. [Finger encoders](http://wiki.icub.org/wiki/Software_interface:_standard_port_names_for_hardware_devices#Finger_encoders)

pattacini, 2016-10-29:
Upon https://github.com/robotology/icub-main/commit/22d86bf642aa0d69619fae7eb94a2487b77c30a8 it is now possible to streamline the above snippets in:

```cpp
// imagine you have available here the Vector encs
// containing the motor encoders and the Vector analogs
// containing the joint encoders

finger.getChainJoints(encs,analogs,joints);
yarp::sig::Matrix tipFrame=finger.getH((M_PI/180.0)*joints);
```

pattacini, 2016-10-29:
cc @nawidj @giuliavezzani

pattacini, 2017-02-11:
It is likely that analogs readings are not covering the full range of motion of the fingers. In this case, the user can proceed by noting down the values of analogs when the fingers are fully extended and fully bent and construct the corresponding matrix of analogs bounds as below, with the first column accounting for full extension and the second for full bending:

``` cpp
yarp::sig::Matrix analogsBounds(15,2);
analogsBounds(0,0)=250.0; analogsBounds(0,1)=11.0; 
analogsBounds(1,0)=233.0; analogsBounds(1,1)=21.0;
// ... 
analogsBounds(14,0)=201.0; analogsBounds(14,1)=12.0;

finger.getChainJoints(encs,analogs,joints,analogsBounds);
yarp::sig::Matrix tipFrame=finger.getH((M_PI/180.0)*joints);
```

cc @giuliavezzani @claudiofantacci @vtikha @vicentepedro

giuliavezzani, 2017-02-12:
Thanks @pattacini  for the improvement! :smiley:  
I tested the new function on the `iCubDarmstadt01` and I got **good positions** both when I passed a `analogBounds` Matrix and when I used the default values in the `getChainJoints()`. :+1:

However, it seems there is still some **problems** in getting fingers positions when the fingers are holding an object, thus, executing **pressure on the object surface**. I experienced this problem on the `iCubDarmstadt01`, whereas I don't remember any similar behaviour on the `iCubGenova01` (but I should test it again).
If you are interested in the problem, you can find more info in [this comment](https://github.com/tacman-fp7/handover/issues/31#issuecomment-279248877).

closed 2020-12-08 by pattacini