robotology/community #153: How to know the sign of MotorEncoders

hu-yue, 2016-10-28:
Hi,

I wanted to ask if there's a way to know the sign of the `MotorEncoders`, as they have different signs in some joints? For the time being I am comparing the readings from `MotorEncoders` and `Encoders` to discriminate this, but is there a function that "returns" the sign or something like that?

Thanks!

ale-git, 2016-10-28:
Hi @hu-yue , in the robots-configuration/yourrobot/hardware/mechanicals/*mec.xml configuration files there is a row with sign and resolutions of motor encoders: 
<param name="RotorEncoderRes">   -14400.0      -14400.0      -14400.0       -14400.0   </param>

Hope this helps.

Alessandro

hu-yue, 2016-10-28:
@ale-git thanks for the reply! This is indeed helpful, so I guess to make the software more "flexible" one should create a parameters file from which the signs should be read (or use the data from `Encoders` as comparison as I'm doing), as it seems then that this information cannot be retrieved on the go from the robot, and as for my knowledge, the signs seem be to different from iCub to iCub.

pattacini, 2016-10-28:
A somewhat related issue https://github.com/robotology/yarp/issues/430.

pattacini, 2016-10-31:
Hi @hu-yue 

I don't think the further level of flexibility you're suggesting will ever help. The user has always the chance to retrieve those info from the same files loaded by `yarprobotinterface`. Although I agree it's not handy, it wouldn't be handier with yet another conf file.

To me it should be more convenient to expose these data through the `API` as per the request I referenced above (but only if they're deemed to be really necessary).

By the way, those signs come from a chosen convention and from how the tendons have been winded or sensors mounted, in the end. In this respect, I know that @randaz81 is trying to push to get all those values consistent among each other over different robots. This would eventually mean a beautiful **`+`**  replicated everywhere and the need for additional care in mounting/maintaining the equipment.

hu-yue, 2016-11-02:
@pattacini I agree that it probably won't help, that was just an idea for my own software, not a suggestion to add it in general. I was just wondering if there was something like this implemented but not documented, as often happens. I will close the issue as for me the question is clear now.

randaz81, 2016-11-02:
Hi @hu-yue,
what is your exact application? Your question sounds strange to me, because I never had the need to expose it through an API, even in the system tests. So it is possibile that there is another way to do what you want to do, without knowing the sign of the motor encoder (e.g. using the gearbox value which is exposed by the API?)

hu-yue, 2016-11-02:
@randaz81 do the gearbox values have signs? I want to control the joint by using data from the motor encoders and have realised that there is a difference in sign. I admit that I just divided by 100 because I knew that it's 100 and did not read the value from the gearbox as I did in other applications. If there's the sign taken into account in the value returned from the `iMotor:getGearBoxRatio` then there's no need of checking for me.

randaz81, 2016-11-02:
yes, gearboxratio is signed such as: motor*gear=joint

hu-yue, 2016-11-02:
@randaz81 ok perfect thanks =)

closed 2020-12-08 by pattacini