robotology/community #154: Factor of 1.5 between errors from iPidControl::getErrors and actual errors

hu-yue, 2016-10-28:
Hi,

I recorded the data from `iPidControl::getErrors` and then also data from `iEncoders`, then I computed the errors by `qdes-iEncoders`. I have realised that the errors from `iPidControl::getErrors` are of a factor of circa `1.5` smaller than the ones I computed. By dividing `qdes-iEncoders` by `1.5` they are basically the same as the ones from `iPidControl::getErrors`.
See plots below:
In blue errors from `iPidControl::getErrors` and in red `qdes-iEncoders`:
[image]

In blue errors from `iPidControl::getErrors` and in red `(qdes-iEncoders)/1.5`:
[image]

I thought that the errors from `iPidControl::getErrors` are the tracking errors, as this is also what he `controlBoardDumper` uses to return the position errors.
The questions are: am I doing something wrong? Or is this factor really there? And if yes, what is it and why?

Thanks!

traversaro, 2016-10-28:
Hi @hu-yue , 
   on which joints did you get this data?

hu-yue, 2016-10-28:
@traversaro these specific plots are for the left ankle roll, however I get the same also on left knee and right knee, I can also test other joints.

hu-yue, 2016-10-28:
It seems to be true for all joints, however on the torso the tracking is quite bad, so I'm not sure about the outcome.

hu-yue, 2016-11-02:
I tried to understand if this was due to the latest firmware and software update, since I had data recorded from before. I have to say that I couldn't actually understand it. The computed errors and the errors from `iPidControl::getErrors` were different, but by dividing the computed errors by 1.5 they did not match with `iPidControl::getErrors`.

In general, I'd like to go back to my initial questions:

- Why are these errors different? Aren't they supposed to be the same or I'm not understanding what are the values from `iPidControl::getErrors`?

randaz81, 2016-11-02:
afaik, they should be identical on all joints of your robot. Suspected bug. @ale-git

hu-yue, 2016-11-02:
@randaz81 @ale-git ok, I wait for further updates then

@KevinAStein FYI

ale-git, 2016-11-02:
Hi @hu-yue , which controlmode has been used for the test? In the plots, is the x axis the time in milliseconds? The errors are measured in degrees? Thanks.

hu-yue, 2016-11-03:
@ale-git 
- the control mode is `position direct`
- the x axis is the number of samples, but you can also say it's time in ms since I was sending commands with a thread rate of 10 ms
- the errors are in degrees

hu-yue, 2016-11-03:
Ok sorry the statement about the time is clearly wrong, total time is 6 sec, so the x axis is just number of samples

ale-git, 2016-11-03:
Hi @hu-yue , you should consider that there is a transmission delay Tt between your qdes at high level and the qdes_pid inside the pid controller, so qdes - iEncoders is in general different from iPidControl::getErrors = qdes_pid - iEncoders. If the delay between the qdes_pid reference and the joint position is affected by a Tc delay related to the pid time constant, then you have:
qdes = A * sin((2*pi/6) * t)
qdes_pid = A * sin((2*pi/6) * (t-Tt))
qenc = A * sin((2*pi/6) * (t-Tt-Tc))
Thus
qdes - qenc  =(about) A * (2*pi/6) * (Tt + Tc) * cos((2*pi/6) * t)  (as shown by the plot)
qdes_pid - qenc =(about) A * (2*pi/6) * Tc * cos((2*pi/6) * t)  (as shown by the plot)
and finally 
(qdes - qenc) / (qdes_pid - qenc) = (Tt + Tc) / Tc 
Given your software configuration, can the transmission delay explain the gain factor?

ale-git, 2016-11-03:
If the above hypothesis is correct, you should read a difference between qdes and iPidControl::getReference.

pattacini, 2016-11-03:
Hi @hu-yue @ale-git 

The acquisitions seem to refer to a steady-state conditions where the error is quite low (~`1-2 deg`).
I would suggest to repeat the experiment logging all the involved quantities during a reasonable transition of `qdes`.

This should allow us to better test the hypothesis and appreciate possible latency among these variables.

ale-git, 2016-11-04:
Just an example:

[image]

hu-yue, 2016-11-04:
@ale-git @pattacini thanks for the replies!

I run another experiment on the left knee joint performing a cos wave that moved the joint from `0` to `-90` and back in `2 [sec]`. I think the transmission delay can explain the difference between the two errors.
The data from `iPidControl::getReferences` seem to be delayed respect to `qDes` and then the encoders have a higher delay time.

Here are plots of the case I tested:

Reference trajectory, encoders and `iPidControl::getReferences`
[image]

Errors:
[image]

Then I suppose that the actual execution errors should be the ones returned by `iPidControl::getErrors` and one should not compute `qDes - iEncoders`, as this would increase the errors due to the delay factor?

ale-git, 2016-11-04:
Hi @hu-yue, yes, the error you should look at is returned by iPidControl::getErrors because it is synchronized to the encoder data. Using qDes - iEncoders introduces a gain factor due to the delay between the qDes and the low level joint position reference.
Cheers,
Alessandro

hu-yue, 2016-11-04:
@ale-git ok perfect, thanks for all the explanations! Now things are clearer.

pattacini, 2016-11-05:
Thanks @ale-git for the great handling of this issue 👍 

@hu-yue I'd just like to add a brief comment upon closure from a bit higher standpoint.
Taking `qdes-enc` as a measure of the error remains perfectly viable though, given that controllers can be easily designed to account for latency too, of course depending on the requirements the designer has in mind.

randaz81, 2016-11-22:
iPidControl::getErrors() has been just implemented in gazebo, see https://github.com/robotology/gazebo-yarp-plugins/issues/260

closed 2020-12-15 by pattacini