robotology/community #155: icub_with_hands model not being "stiff"

emilio-cartoni, 2016-11-07:
Hi,
we are starting to use Gazebo for iCub simulations.
In particular, we would like to use the icub_with_hands model.
We tried to make the iCub execute some movements using the yarpmotorgui and we noticed that even if we are using Position Control and Stiff mode, the iCub moves joints that were given no orders (as if it is trying to balance some gravity/force).
This does not happen with the icub model (no hands).
Why is that so?

traversaro, 2016-11-08:
Hi @emilio-cartoni , 
  I strongly suspect this is due to some gains of the joint-level PID controllers not properly tuned. 
  In particular on which joint did you see this behavior?

emilio-cartoni, 2016-11-08:
Hi @traversaro ,
we were mostly using the first three joints of the left arm.
By browsing through the .sdf files it seemed to me that /icub/icub.sdf and /icub_with_hands/icub_with_hands.sdf refer to the same configuration files (i.e. /icub/conf/gazebo_icub_torso.ini) and that they would use the same PIDs (but maybe I am wrong, I am not familiar with Gazebo).
Even where they use different .ini files, i.e. gazebo_icub_left_arm_no_hand.ini vs gazebo_icub_left_arm_no_hand_for_no_hand_model.ini the PiD values in the files look the same (sections [GAZEBO_PIDS], [GAZEBO_VELOCITY_PIDS]).

Here is a small example of what we do:
1) Launching gazebo with icub.world

2) Delete the iCub model from the scene
3) Insert iCub_with_hands model
4) start yarpmotorgui
5) Set joint 0 of left_arm (l_shoulder_pitch) velocity to 90.0
6) Set joint 0 of left_arm (l_shoulder_pitch) position to 5
7) Set joint 0 of left_arm (l_shoulder_pitch) position to -95
The wide and fast swing from one extreme position to the other will cause many joints to change their positions, including torso, elbow and wrist joints.
If the iCub is not fixed to the ground it will fall.

8) Delete the iCub_with_hands model from the scene and close the motorgui.
9) Insert iCub (no hands)
10) Repeat 5-7 on this model.
The iCub will slightly oscillate on the torso but it will stay mostly rigid (and it will not fall).

randaz81, 2016-11-08:
I can easily guess that the inertial parameters of the forearm/hand models are excessively large (graphically inspected within the simulator). This will be fixed.
However, I suggest not to move the l_shoulder_pitch joint of the REAL robot at a speed of 90 deg/s, it's dangerous for the robot and for surrounding environment and the humans nearby!

randaz81, 2016-11-08:
commit commit 608313507185ae62d8bdafab215587ee14c3e440 should have fixed the inertia matrix of the fingers. Can you give a try?

emilio-cartoni, 2016-11-08:
@randaz81 Thanks, it is definitely improved: the torso oscillates a lot less and the elbow is stiff.
The only problematic joint is the wrist (prosup, pitch, yaw).
It still seems that the hand is too heavy for those joints to keep it in position.
Joint wrist_yaw is also affected by this bug robotology/icub-gazebo#35  (but I am not sure if it contributes to this behavior).

randaz81, 2016-11-10:
robotology/icub-gazebo@a5205fa fixes the effort limits. It should be fine now. Please fell free to open a new issue in the **`icub-gazebo`** repo if you notice further issues. Thank you!

emilio-cartoni, 2016-11-16:
Thanks @randaz81, the body is now fine (torso, elbow, wrist).

There is some side-effect of the fixes on the hands which now have a constant small tremor and sometimes a finger twitches for a split second - I'll open a separate issue in the icub-gazebo later as suggested.

Yeshasvitvs, 2016-11-16:
I noticed the finger twitching too in my machine @randaz81

closed 2020-12-09 by pattacini