robotology/community #156: Using Lua as a state machine

amaroyo, 2016-11-08:
Hey guys! @vtikha 

Are you still using Lua in yarp? I wanted to create a state machine in which, in different states it could send and receive information/commands from different modules. Do you recommend me to use Lua? Or is there something else which is better?

Thanks a lot!
Alex.

pattacini, 2016-11-08:
Hi @amaroyo 

Yep, I can confirm that using `Lua` is still our preferable way to code not only **state machines** but also **behavior based applications** within our infrastructure.

cc @vtikha @apaikan

jgvictores, 2016-11-08:
Hi @amaroyo @pattacini @vtikha @apaikan  !!

Disclaimer: no intention to go against any recommended best practice! Will begin `Thinking in Lua` asap! :wink: 

The [robotic video-game](https://github.com/asrob-uc3m/robotDevastation) I'm involved in is currently written in C/C++. Facts are we actually made a small [state machine](https://github.com/asrob-uc3m/robotDevastation/tree/be4a3aa21c74ed660f794c3c663f3cdb6c1cee15/src/libraries/StateMachineLib) library for it (with [some limited documentation](http://asrob.uc3m.es/index.php/Robot_devastation_FSM) and [tests](https://github.com/asrob-uc3m/robotDevastation/tree/develop/test)), and are going to present it at an [upcoming workshop](https://roboticslab-uc3m.github.io/workshop-humanoids2016/). Feel free to take a look, reuse, and/or file an [issue](https://github.com/asrob-uc3m/robotDevastation/issues) if you see anything worthy of comment. Regarding code integration (for not having to install the video-game as a dependency), I've spoken with other project leaders and we have the intention of splitting several projects into mini-repositories and then unifying them through git submodules. We could give priority to this library if required.

pattacini, 2016-11-08:
No problem @jgvictores, you're very welcome to advertise new solutions you're working on 😄.

##### Side note
Aiming to mini-repositories is a good way of organizing code and its dependencies, but using `git submodules` is discouraged, at least in our view; see discussion in https://github.com/robotology/icub-main/issues/360.

By the way, why don't you try our [**superbuild**](https://github.com/robotology/superbuild-example) mechanism instead?

cc @drdanz

jgvictores, 2016-11-08:
@pattacini I was actually quite reluctant to `git submodules`, which is in part why we haven't been doing it more. Thanks for the pointer to the superbuild mechanism, I'll comment on there if I have any doubts!!

pattacini, 2016-11-08:
@jgvictores that's good!

@amaroyo I've found this resource online: http://wiki.icub.org/images/4/46/Rfsm_yarp.pdf.
@apaikan do you have better locations where to point at?

apaikan, 2016-11-09:
@pattacini: thanks for sharing the tutorial. I guess that is the only tutorial using yarp with rfsm
@amaroyo: Honestly, I do not see any special need for other doc or tutorials regarding rFSM and Lua. What actually one needs is just to know 

- how to compile yarp-lua binding (http://www.yarp.it/yarp_swig.html#yarp_swig_lua) and use yarp from Lua (https://github.com/robotology/yarp/blob/master/bindings/example2.lua)
- how to write state machine using rFSM (https://people.mech.kuleuven.be/~bruyninc/rFSM/doc/README.html)

p.s. @jgvictores : There are plenty reasons why we decided to stay away from c++-implemented or pure xml-based state machines (e.g., SCXML). Using scripting for developing state machines for component coordination has distinct advantages as thoroughly discussed in this article: http://joser.unibg.it/index.php?journal=joser&page=article&op=view&path%5B%5D=52.

amaroyo, 2016-11-09:
@jgvictores Thanks, I will take a look to the code!

@pattacini Yes! I found the same slides in google, but I saw that the date was 2013 :)

@apaikan Thanks for the documentation, I will take them a look and tell you how it goes!

closed 2020-12-08 by pattacini