robotology/community #158: iCub Eye - How to get eye pose from encoder values

claudiofantacci, 2016-11-15:
Hi everyone,
this a small, self-contained, "tutorial" on how to jointly use [iCub joint values](http://wiki.icub.org/wiki/ICub_joints) with [`iCubEye`](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubEye.html) class to get, either left or right one, [eye pose](http://wiki.icub.org/wiki/ICubHeadV2Kinematics) ✏️ 
[image]

Assume that you store/read encoder values from the iCub torso and head through, respectively, `/icub/torso/state:o` and `/icub/head/state:o`: you will find yourself with **3 values for the torso** and **6 values for the head** 🤔

According to the iCub reference frames (click on the following image for details)
[image]
you have 3 joints for the torso, and three for the neck.
So what are the other 3 values found reading head encoders? 🤔
They are, of course, related to the eyes!
[image]

Let's see what they are: [_vergence_, _version_ and _disparity_](http://wiki.icub.org/wiki/Vergence,_Version_and_Disparity)!
The following image depicts _version_ (Vs) and _vergence_ (Vg) which are of use for our purposes.
[image]

To get the an eye pose, we need to _feed_ 🍴 the `iCubEye` interface with the encoders from the ROOT to the desired eye, plus the L or R value (see previous picture) if you either need the left or the right eye pose.
As a result, the values to be passed to `iCubEye` are:

1-3. **3 encoders from the torso** (the torso encoders from `/icub/torso/state:o`)
4-6. **3 encoders from the neck** (the first three values from `/icub/head/state:o`)
7. **1 encoder for the eye tilt** (the fourth value from `/icub/head/state:o`)
8. **L|R value for the left|right eye** (which is to be evaluated from the fifth and sixth values of `/icub/head/state:o`)

The L|R value can be evaluated from the version and vergence following the linear relationship reported [here](http://wiki.icub.org/wiki/Vergence,_Version_and_Disparity):
```
L = Vs + Vg/2;
R = Vs - Vg/2;
```

We now have all the values we need! We are ready to use [`iCubEye`](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubEye.html) class to get the desired eye pose.
Here is a (almost) ready-to-use code snippet:
```c++
Vector q(8);
...
/* Store the 8 encoder values described before in q. */
...
iCubEye icub_kin_eye("left_v2");
icub_kin_eye.setAllConstraints(false);
icub_kin_eye.releaseLink(0);
icub_kin_eye.releaseLink(1);
icub_kin_eye.releaseLink(2);
Vector eye_pose = icub_kin_eye.EndEffPose(CTRL_DEG2RAD * q);
```

Let's now dig into the detail!

First and foremost: `iCubEye` class is fully documented [here](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubEye.html) ✌️ 

```c++
setAllConstraints(false)
```
releases the constraint of all chain links so that, even if any of the encoder goes beyond the default encoder constrained value, the encoders are not saturated to their constraints. If you feel uncomfortable with this, you can use [`alignJointsBounds`](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubEye.html#a7c99fcd586adb0ecdda1662ce5e4d6c3) member function to properly set the encoder limits.

```c++
releaseLink([0|1|2])
```
releases the first three links of the chain from the ROOT to the eye, i.e. the 3 DOF of the torso. By doing so, we are telling the interface to use the first 3 values of vector `q`
 as torso encoders.
⚠️ Beware ⚠️ : by not releasing the first three links while having used them, will most likely produce wrong/unwanted results!

```c++
EndEffPose(CTRL_DEG2RAD * q)
```
finally gives us the pose of the eye. The output vector is, by default, a 7x1 vector with `xyz` coordinates plus a 4-component axis/angle vector. If you prefer RPY angles, you just need to provide `false` as a second argument. Documentation of the member function can be found [here](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iKinChain.html#a6d5704c9f3aaeed61ea03f9289223584).
⚠️ Beware ⚠️ : a common pitfall is to forget to convert the encoder values, expressed in angle degrees, in radians multiplying by `CTRL_DEG2RAD`.

If you went up through here, well, congratulations 🎉 you now have the eye pose at your preferred use!

vicentepedro, 2016-11-15:
Hi Claudio!

Thanks for this "tutorial" ;)

I have an issue related to the torso joints update.

Usually, when I use the iCubArm chain (similar to the iCubEye, I suppose),  I need to update the torso joints in the reverse order.
```
armKin.releaseLink(0);
armKin.releaseLink(1);
armKin.releaseLink(2);
armKin.blockLink(0,CTRL_DEG2RAD*encodersTorso[2]);
armKin.blockLink(1,CTRL_DEG2RAD*encodersTorso[1]);
armKin.blockLink(2,CTRL_DEG2RAD*encodersTorso[0]);
```
Can you confirm that for the iCubEye we don't need to reverse the joint order?

Thanks in advance,
Pedro

claudiofantacci, 2016-11-15:
Hi @vicentepedro :-)

For "historical" reasons, the iCub torso joints are reversed, [see here](http://wiki.icub.org/wiki/ICub_joints#Torso), i.e., (Yaw, Roll, Pitch).
As a result, you read the encoders and then you have to _flip_ them from (Yaw, Roll, Pitch) -> (Pitch, Roll, Yaw).
I updated the tutorial accordingly.

Hope it helps!

vicentepedro, 2016-11-15:
Thanks, @claudiofantacci. Now, I think the tutorial it's perfect.

Hope it helps the iCub community ;)

claudiofantacci, 2016-11-15:
Cool, thanks @vicentepedro !

pattacini, 2016-11-30:
Interestingly, the same recipe is also valid to correctly populate other `iKin` classes, such as [`iCubArm`](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubArm.html) or [`iCubTorso`](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubTorso.html).

claudiofantacci, 2016-11-30:
Should I open another QA, say,
_iCub iKin - How to get end effector pose from encoder values_
with a phrase like yours @pattacini which, however, simply points here ?

pattacini, 2016-11-30:
@claudiofantacci I don't believe it's really necessary. Later searches through GitHub will catch your nice instructions anyway 😉

closed 2020-12-08 by pattacini