robotology/community #161: [gazebo simulator] is it possible to increase the physics max step size?

valeriosperati, 2016-11-23:
I'm newby in Gazebo and running the first tests. I inserted "iCub with hands" in the simulator and run succesfully  the tutorial_arm.  Then I tried to increase  the physics max step size (in the gazebo gui) from 0.001 to 0.002, in order to have a faster (less accurate) simulation, obtaining sometimes a crash of gazebo, sometimes the icub crumpling on itself. 
Is is this behaviour expected?  Is there some way to make the simulation run faster?

pattacini, 2016-11-24:
Senior developers helping on this might be @randaz81 @traversaro @Yeshasvitvs 

@valeriosperati you should have received an invitation to join our GitHub community (this will allow you for example to assign labels). If you did not receive it, please check the email account associated to your GitHub profile.

Thanks @emilio-cartoni for contributing to keep Q&A neat 😉

traversaro, 2016-11-24:
Hi @valeriosperati ,  
  in general all the models provided in the [`icub-gazebo`](https://github.com/robotology/icub-gazebo) repository have been tested with the default Gazebo physics step size of `0.001 s`. Notice that all the Gazebo physics engine at the moment have a fixed-step integrator, so this set the **actual step** of the integrator.

Unfortunately the differential equations describing the evolution of a complex multibody system such as the iCub (with small light links such as the fingers and large heavy links such as the chest) are known to be rather **[stiff](https://en.wikipedia.org/wiki/Stiff_equation)**, so it does not surprise me that you have numerical stability problems by doubling the integrator fixed step. 

Going back to your (I guess) original goal, I think there are several options that you may want to investigate:
* If you do not need the hands or the camera, you can try the `iCub (without hands)` model. 
* If you do not need the camera, you can try to disable the camera plugin or reduce the camera framerate. 
* If you are interested in simulating the iCub attached to a pole, you can use the `iCub (fixed)` model, and thus you avoid to simulate the interaction between the feet and the ground. 
* If you don't need to account for the model collision, you can try to comment out all the `collision` elements from the SDF, and that should reduce the time that the simulator spends in checking for collisions. 
* At the moment we do not provide a model without the legs, but that could also improve the performances. If you are confident with the SDF format, you can look into removing manually the links and the joints. 

In general the right thing to do before taking any of these steps would be to profile Gazebo to understand which aspect of the simulation is your bottleneck, but unfortunately this require to compile Gazebo from source and some insight on the internals of Gazebo. If you are nevertheless interested in this, you can look into [1].

[1] http://answers.gazebosim.org/question/11998/why-does-performance-decrease-when-removing-inserting-models/?answer=12009#post-id-12009

randaz81, 2016-11-24:
It's possible, in principle. I do no know about gazebo crashes (sometimes they occur without a specific reason due to problems in ogre, the graphic library). Instead, if you see the icub crumpling on itself, it is most likely that current PID gains of icub are not stable anymore. Indeed, they are tuned to work at a specific simulation step and they should be reduced accordingly if your increase the step size (while decreasing it is always ok).

Unfortunately, it is difficult to suggest a solution for your problem. 
Obviously my first suggestion is to use a more powerful machine. I am able to run complete icub simulation near realtime on my laptop, which is a good laptop, but not a huge machine. (*)
Alternatively you can remove from the SDF the collision tags, this will speed up the simulation a lot. You can also remove cameras if you are interested in body simulation only. 
Finally, I noticed that on machines with old graphic cards, the physics computation is ok, but the display is running very slow. For these machines I suggest NOT to run fullscreen and to reduce the size of the gazebo window as much as possibile. This is enough to increase the FPS number displayed in bottom right corner.  

(*) you can also run gzserver (the physics simulation) on a remote machine, and gzclient (the display) on you pc.

valeriosperati, 2016-11-25:
Thanks a lot for the explanations and the informations.  Right now I'm just exploring Gazebo, testing the icub tutorials both on Gazebo and in iCub_SIM.  For now, I will keep the  physics step size to its default value. 

PS: please, it is the first time that I  interact with GitHub issues: can we consider this issue closed? Is it up to me to "close and comment"? Thanks, Valerio

pattacini, 2016-11-25:
Hi @valeriosperati 

Here's some info on [how to contribute](https://github.com/robotology/QA/blob/master/.github/CONTRIBUTING.md).

Usually, the OP (you in this case) is responsible for judging whether the issue can be closed or not. Maintainers can also kick in and say a word in this respect.

I think we can close it.

closed 2020-12-09 by pattacini