robotology/community #165: [announcement] YARP 2.3.68 Released

drdanz, 2016-11-28:
YARP 2.3.68 was released today and is now available for download at: https://github.com/robotology/yarp/releases/. This release is a _feature_ release and introduces several new features and important changes since YARP 2.3.66.

A (partial) list of bug fixed and issues resolved in this release can be found
[here](https://github.com/robotology/yarp/issues?q=label%3A%22Fixed+in%3A+YARP+2.3.68%22).

Important Changes
-----------------

### Dependencies

* Readline library is no longer used and was replaced with the NetBSD Editline
  library (libedit-dev on debian/ubuntu).
* [YCM](http://robotology.github.io/ycm) 0.2.2 is now an optional dependency.

### YARP_OS

* The yarp::os::Contact class was widely refactored:
  * The add*() methods are now deprecated. The new set*() methods should be used
    instead.
  * The by*(), invalid() and empty() factory methods are now deprecated. The
    class constructors should be used instead (a few new constructors were added
    to allow this).
  * The byConfig() method is now deprecated. fromConfig() should be used
    instead.
  * The Contact::toURI() method now accepts an includeCarrier parameter (true by
    default) to simplify the generation of the URI without the carrier.
    Moreover the string returned always end with a `/` character.
  * ROS integration was heavily improved and it is now possible to register
    one `Publisher` and one `Subscriber` on the same `Node` using the same
    topic, and to use tools like `roswtf` and `rqt`.
    If you are using YARP with ROS, upgrading is strongly recommended.
  * Added `ConnectionReader::setParentConnectionReader` method. This method
    should be used by `Carriers` that reimplement `modifyIncomingData` and that
    return a ConnectionReader that is not the same as the one in input.
* `PortCore` now uses the "new" version of the protocol for signalling to
  suppress the reply. This was expected since YARP 2.1.2, but it was never
  applied. Both versions of the protocol are accepted in input, unless
  `YARP_NO_DEPRECATED` is enabled, in this case YARP it is not compatible with
  previous releases since only the new version is recognized.
* It is now possible to get and set parameters to portmonitors directly attached
  to a port.

### YARP_sig

* Important: we removed methods for interoperability with the GNU Scientific
  Library (GSL).
  The reason for this was to avoid licensing problems generated by the inclusion
  of a structure defined by GSL. Methods for interoperability have been
  implemented in a new library libYARP_gsl (see below). In details:
  * We removed yarp::sig::Matrix::getGslMatrix()
  * We removed yarp::sig::Vector::getGslVector()

### YARP_dev

* New depth camera stack, device driver interface and wrapper. The old
  interfaces and devices will be deprecated in the next release, when all the
  features will be ported to the new stack.

### YARP_gsl

* This library contains methods for interoperability with the GNU Scientific
  Library. Notice that this library is released as GPL (and not LGPL) because it
  uses a structure defined by GSL.
  * Checkout yarp::gsl::GslMatrix and yarp::gsl::GslVector

### YARP_math

* libYARP_math can be compiled with GSL or Eigen (via CMake flag
  `CREATE_LIB_MATH_USING_GSL`). Important: the use of GSL and this flag will be
  removed in the next release of YARP.
[... 186 lines cut ...]
git history using `git shortlog -ens --no-merges v2.3.68..v2.3.68`):

```
   430  Daniele E. Domenichelli <daniele.domenichelli@iit.it>
    49  Andrea Ruzzenenti <andrea.ruzzenenti@iit.it>
    47  Marco Randazzo <marco.randazzo@iit.it>
    44  Alberto Cardellino <Alberto.Cardellino@iit.it>
    41  Silvio Traversaro <silvio.traversaro@iit.it>
    34  Tobias Fischer <t.fischer@imperial.ac.uk>
    16  Francesco Romano <francesco.romano@iit.it>
    13  Claudio Fantacci <claudio.fantacci@iit.it>
    12  Alberto Cardellino <alberto.cardellino@iit.it>
    10  Lorenzo Natale <lorenzo.natale@iit.it>
    10  Matteo Brunettini <matteo.brunettini@iit.it>
     9  Ugo Pattacini <ugo.pattacini@iit.it>
     8  Vadim Tikhanoff <vadim.tikhanoff@iit.it>
     7  Alexandros Paraschos <paraschos@ias.tu-darmstadt.de>
     6  Ali Paikan <ali.paikan@iit.it>
     3  Giovanni Saponaro <gsaponaro@isr.ist.utl.pt>
     1  Claudia Latella <claudia.latella.86@gmail.com>
     1  Gabriele Nava <gabriele.nava@mail.polimi.it>
     1  Plinio Moreno <plinio@isr.tecnico.ulisboa.pt>
     1  Valentina Gaggero <valentina.gaggero@iit.it>
     1  YARP Developers <yarp0-devel@lists.sourceforge.net>
```

drdanz, 2016-11-28:
cc @robotology/all-robotology-developers

traversaro, 2016-11-28:
@robotology/all-robotology-developers If after upgrading YARP you are not able to compile icub-main, please check https://github.com/robotology/icub-main/issues/389 .

Tobias-Fischer, 2016-12-02:
Hi @drdanz @mbrunettini,
Will this release be made available on the repository (http://www.icub.org/ubuntu)?

Thanks, Tobi

drdanz, 2016-12-02:
@Tobias-Fischer yes, @mbrunettini is working on it, the packages will be available as soon as possible.

closed 2020-12-12 by pattacini