robotology/community #166: Desactivate `CUDA` in cmake

pattacini, 2016-12-01:
@gregoire-pointeau commented on [Thu Dec 01 2016](https://github.com/robotology/yarp/issues/981)

Hello, we are working on the cluster. We have the blades and the cuda machine that share the same sources.

While compiling `yarp` then `icub-main`, it works on the `cuda-machine` but on the blades we have a problem of cuda:

This is the `cmake`:
>CMAKE_BUILD_TYPE                 Release                                                               
 CMAKE_INSTALL_PREFIX             /usr/local                                                            
 CREATE_GUIS                      OFF                                                                   
 CREATE_LIB_MATH                  OFF                                                                   
 CREATE_SHARED_LIBRARY            ON                                                                    
 CREATE_YARPDATADUMPER            ON                                                                    
 CREATE_YARPMANAGER_CONSOLE       ON                                                                    
 CREATE_YARPROBOTINTERFACE        ON                                                                    
 CUDA_BUILD_CUBIN                 OFF                                                                   
 CUDA_BUILD_EMULATION             OFF                                                                   
 CUDA_SDK_ROOT_DIR                CUDA_SDK_ROOT_DIR-NOTFOUND                                            
 CUDA_TOOLKIT_ROOT_DIR            /usr/local/cuda                                                       
 CUDA_VERBOSE_BUILD               OFF                                                                   
 ENABLE_DASHBOARD_SUBMIT          OFF                                                                   
 OpenCV_DIR                       /usr/local/src/robot/opencv/build                                     
 YCM_DIR                          YCM_DIR-NOTFOUND                                                      

and this is the output:
>  CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:97 (MESSAGE):
   Could NOT find CUDA: Found unsuitable version "7.5", but required is exact
   version "4.2" (found /usr/local/cuda)
 Call Stack (most recent call first):
   /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:286 (_FPHSA_FAILURE_MESSAGE)
   /usr/share/cmake-2.8/Modules/FindCUDA.cmake:723 (find_package_handle_standard_args)
   /usr/local/src/robot/opencv/build/OpenCVConfig.cmake:48 (find_package)
   /usr/local/src/robot/opencv/build/OpenCVConfig.cmake:276 (find_host_package)
   cmake/ycm-0.2/find-modules/FindOpenCV.cmake:88 (find_package)
   cmake/YarpFindDependencies.cmake:363 (find_package)
   CMakeLists.txt:62 (include)

Is there a way to deactivate `cuda`?

Thank you

@AnneLaureM

pattacini, 2016-12-01:
Hi @gregoire-pointeau 

`cuda` and the `blades` should have the same OS, but of course `cuda` has its own stuff related to the GPU platform. Therefore, compilation should always go neatly on `cuda` but might bring about troubles on `blades`.

However, the same binaries compiled on `cuda` should run smoothly also if launched from `blades`, because they basically share the same OS. Thus, you wouldn't need to compile code on `blades`.

In case you really need to compile on `blades`, one possible workaround is to keep sharing sources, but differentiate the build process. This would entail to tailor e.g. the `YARP_DIR` and `ICUB_DIR` variables specifically for the `blades`.

@mbrunettini @vtikha am I wrong or missing anything?

gregoire-pointeau, 2016-12-01:
On the blades were always set 2 directories in each project (`yarp`, `icub-main`...):
- `build-cuda`   (to be executed on the `icub-cuda`)
- build-x86_64   (to be executed on the blades)

vtikha, 2016-12-01:
@gregoire-pointeau can you confirm that the `CUDA` and the `blades` have the same OS?

pattacini, 2016-12-01:
Nonetheless, if they have two separate build directories, then you shouldn't experience any problem, in principle.

Are the env variables correctly set up? What are the values of `YARP_DIR` and `ICUB_DIR`, for both the systems? Perhaps, you could do `which yarp` on `blades` and check out that the path is the intended one.

Did you try out purging the cmake cache?

gregoire-pointeau, 2016-12-02:
@AnneLaureM  Could you have a look please? I'm never comfortable with everything that is architecture-related...

drdanz, 2016-12-02:
@gregoire-pointeau did you build OpenCV manually? Apparently the warning comes from inside the `OpenCVConfig.cmake` file...

vtikha, 2016-12-02:
@gregoire-pointeau let me add something to @drdanz comment, if `opencv` has been built with `cuda` capabilities you will need to have two separate `build` directories for it to run on the `blades`.

pattacini, 2016-12-05:
Hi @gregoire-pointeau and @AnneLaureM 
Any news?

AnneLaureM, 2016-12-15:
Hi everybody,

I am REALLY SORRY for the response time (I had my defense thesis on monday :-) ) . 
A few months ago, we have compiled opencv 3.0  and by default opencv has some dependencies with cuda, so by default the cuda system librairies (in our case cuda 4.2 on Debian Wheezy...). So finally, I have just removed cuda 4.2, recompiled everything with cuda 7.0 and everything seems ok now. 
Sorry for the ticket, The problem seemed rather clear with the cmake error. I did not pay enough attention to the error... :-(

I just have one last question... Which operating system do you use on your blades and the cuda machine at IIT ? Would it not be preferable to upgrade the whole cluster with Debian Jessie ? Is there a procedure to do it on the blades ?

Thanks very much for your help !!

Cheers,

Anne-Laure

pattacini, 2016-12-15:
Congratualtions @AnneLaureM, you made the grade 🎉 🎓 😄 

Actually, we've lately upgraded our system to the use of **Ubuntu Xenial**.

It's likely that @mbrunettini can point you at a list of useful instructions he might have for that.

AnneLaureM, 2016-12-21:
Hi  !!

Thanks @pattacini :-) Thanks to the grade, I am now able to read properly cmake errors :-D :-D.
I am surprised that the upgrade has been made with Ubuntu :-( !! @mbrunettini can you send us the list of instructions ? Do you think it is possible to upgrade the whole cluster into Debian Jessie ?

Cheers,

Anne-Laure

pattacini, 2017-01-12:
@AnneLaureM is this still active?

closed 2020-12-15 by pattacini