robotology/community #167: Request to update icub-main and robots-configuration to address iCub Head 2.5

pattacini, 2016-12-01:
Dear @robotology/all-robotology-developers 

Since in our latest mechanics of the iCub head the IMU is displaced wrt its standard location, we have been forced to update the way we handle hardware versioning in order to better consider future releases.

In particular, we have:
- Extended **`iKin`** to define the new `2.5` head mechanics.
- Streamlined **`iKinGazeCtrl`** to deal with multiple hardware versions (i.e. `1.0`, `2.0`, `2.5`).
    The `--headV2` command-line option has been superseded by **`--head_version x.y`**.

As a result, we would kindly ask you to update your [**`icub-main`**](https://github.com/robotology/icub-main)`/ master` and [**`robots-configuration`**](https://github.com/robotology/robots-configuration)`/ master` local repositories.

Furthermore, you should also take care of updating robot configuration files possibly customized on your system (i.e. those located under `./local/share/yarp/robots/...`), which are currently shadowing their counterparts with default values (i.e. those located in `$ICUB_DIR/share/robots/...` or in `/usr/local/share/yarp/robots/...`).

Related resources:
- https://github.com/robotology/icub-main/issues/285
- https://github.com/robotology/icub-main/pull/391
- https://github.com/robotology/robots-configuration/pull/13

Tobias-Fischer, 2016-12-01:
Hi @pattacini,
I think we have the head 2.0 here on `iCubLondon01`. Could you please confirm this? If this is the case, we should update the files in `robots-configurations` for our robot as well.

Thanks a lot,
Tobi

pattacini, 2016-12-01:
Hi @Tobias-Fischer,

What I did in https://github.com/robotology/robots-configuration/pull/13 was to search for `--headV2` and replace it with the new option. I didn't find anything for `iCubLondon01`, hence - presumably - from `robots-configuration` standpoint you've got an **head 1.0**.

Can you drop here a picture of the iCub neck? 😄

Tobias-Fischer, 2016-12-01:
Hi @pattacini, okay, I see. Please find below the pics, I hope you can see what you need to see ..
[image]
[image]
[image]

traversaro, 2016-12-01:
That's definitely a **V2** neck.

pattacini, 2016-12-01:
Yed, indeed! Thanks @Tobias-Fischer for the pictures.

pattacini, 2016-12-01:
@Tobias-Fischer can we open an new issue on this?

I'm asking because I'm afraid there're other files to be updated (see the [calibrators](https://github.com/robotology/robots-configuration/blob/master/iCubGenova01/calibrators/head_calib.xml#L10) of `iCubGenova01` for example).

Tobias-Fischer, 2016-12-01:
Sure, I'll open the issue.

closed 2020-12-12 by pattacini