robotology/community #169: How to add the saccadic motion to Vizzy's Gaze controller

plinioMoreno, 2016-12-02:
@pattacini,

We want to add the saccadic gaze motion to Vizzy. What are the requirements in order to do this?

pattacini, 2016-12-04:
Hi @plinioMoreno ,

The **saccadic behavior** is described in section **D.3** of "[_A Cartesian 6-DoF Gaze Controller for Humanoid Robots_](http://www.roboticsproceedings.org/rss12/p22.html)".

Saccades are **fast open loop movements** of the eye bulbs that are achieved by sending **position commands** with **very high reference speed**.

Relevant code can be browsed here:
- [Computation of set-points](https://github.com/robotology/icub-main/blob/master/src/modules/iKinGazeCtrl/src/solver.cpp#L302-L366)
- [Transmission of control commands](https://github.com/robotology/icub-main/blob/master/src/modules/iKinGazeCtrl/src/controller.cpp#L437-L476)

Since the eyes are normally controlled in velocity for doing **smooth pursuit**, we need the so-called [**`MIXED`**](https://github.com/robotology/icub-main/blob/master/src/modules/iKinGazeCtrl/src/controller.cpp#L534) control mode, which is capable of accepting at the same time and blending together positions and velocity commands.

How does it work in brief? The position set-points received by the firmware are basically corrected at run-time by integrating the velocity commands. Therefore, aside from developing the saccadic control at high level, you're also required to **implement in the firmware** this special mixed mode.

You can find out more in our [**specifications**](http://wiki.icub.org/images/c/cf/ICub_Control_Modes_1_1.pdf).

**Alternatively**, you might consider starting with a simpler implementation where you can switch to `POSITION` mode to commence a saccade, wait until it comes to a complete stop to then switch back to `VELOCITY` mode. The final effect won't be as nice as if you use `MIXED` mode, but this may represent only a first release.

pattacini, 2016-12-10:
Closing for now.

closed 2020-12-08 by pattacini