robotology/community #172: iCub lift/harness system: where/how to buy/build it ?

nisommer, 2016-12-13:
Hi all,

We recently received our second iCub, this one being able to walk.
Where can I find information on the harness system for safety while it is walking ?
I could only find the video n12 in [](http://wiki.icub.org/wiki/Installation_of_the_robot) to show how to lift the icub, but no information how to build or buy the harness/lift system.

Thank you for your help.
Nicolas

pattacini, 2016-12-13:
Is anyone from @robotology/icub-support-team willing to follow @nisommer's request?

cc @julijenv @Fabrizio69

jbmouret, 2016-12-14:
Hi Nicolas,

do you mean this kind of crane?
http://www2.thuasne.fr/thuasne/front/site/france/cache/offonce/pid/853;jsessionid=A2917B709DCA8B840AA86D3CE61A387D

nisommer, 2016-12-14:
Hi, yes this is what I'm talking about.
I guess it's between this or building our own simple system.
Thank you. I will investigate this solution for now.

jbmouret, 2016-12-15:
FYI, this costs about 700 euros (so, not much compared to a home-made
system).

closed 2020-12-15 by pattacini