robotology/community #173: [Announcement] C++11 and CMake 3.0 Required on YARP and YCM devel

drdanz, 2016-12-15:
We decided to drop support for Debian Wheezy and Ubuntu Trusty (see YARP supported releases policies [[1]](http://wiki.icub.org/wiki/YARP_Supported_Distributions) for details).
Moreover we decided to drop support for Visual Studio 10 2010 and Visual studio 11 2012.

These compilers will still be supported on `master`, but they are no longer supported in `devel` and will not support next major release. Just to clarify, the old compilers are still supported in master.

This means that we are now "free" of old compilers and can (and will) finally start using c++11.

We are also taking the chance for bumping CMake dependency to 3.0 and start using some new features and remove some old hacks and workarounds. If you want to use devel, and you are still using CMake 2.8 please update.

CC: @robotology/all-robotology-developers

Tobias-Fischer, 2016-12-15:
Does that mean that there finally will be a new PC104 image for the iCub? :)

drdanz, 2016-12-15:
@Tobias-Fischer The debian jessie version of the PC104 image is available since may, is there a problem with that image?

Tobias-Fischer, 2016-12-15:
Great! I didn't know this version is available (was there an announcement?), we'll update `iCubLondon01` accordingly :)

traversaro, 2017-07-28:
Just a useful information for people still working on legacy system such as Ubuntu 14.04:
CMake 3.5 is now available in Trusty repository under the `cmake3` package, see https://packages.ubuntu.com/trusty/cmake3 .  

Disclaimer: this does not change the original announcment, i.e. **that Ubuntu 14.04 is not officially supported by the latest version of YARP**.

closed 2020-12-12 by pattacini