robotology/community #175: Possible ways to run different versions of YARP/iCub for legacy demos?*

kt10aan, 2016-12-20:
Ideally, all our software/demos would be updated to be compatible with the latest iCub/YARP version.
However, in practice, we would have some demos that haven't been updated for some time, e.g. for a year or so (for various reasons: researchers leaving, focus on different priorities etc).

We do use gar-installer for having different yarp/iCub builds and dependencies (at least on the ''icubsrv'' machine), in combination with the [''module''](http://modules.sourceforge.net/) package.

However I was thinking if the same thing could be possible with ''YCM'' and I would like to ask if someone has setup a similar system (because that would make the integration/process much easier).

pattacini, 2016-12-20:
@drdanz @traversaro any hints on that?

pattacini, 2017-01-12:
Discussing this doesn't seem much appealing...

Tobias-Fischer, 2017-01-12:
Maybe www.docker.com could help? I'd be interested in that as well.

drdanz, 2017-01-17:
@kt10aan sorry for the delay, I just noticed this issue.

One of the features of YCM is that you can set a `<repo>_TAG` cmake variable for each repository (for example `YARP_TAG`, `ICUB_TAG`, etc.).

This allows you to use a fixed commit hash from each repo.

The idea behind this was to be able to have "snapshots" of a demo at the exact point when you are 100% sure that the demo is working.
This hasn't been used (AFAIK) by anyone for creating "snapshots", but in a similar way to ensure that the developers are all using the same version of the libraries (see [this file](https://github.com/robotology-playground/robotology-superbuild/blob/master/cmake/ProjectsTags.cmake) for example).

There might be some issue if you have more than one installation of the same repo, but there are a few work around for them (ensure that they are not in the `PATH`, `LD_LIBRARY_PATH`, ecc, set the `USE_SYSTEM_<REPO>` variable to `OFF`.

Let me know if you want to try, I  can help you with that.

pattacini, 2017-01-27:
Closing, feel free to open it again.

closed 2020-12-08 by pattacini