robotology/community #176: Sending Video Stream From the C++ Code

LakshithaPW, 2016-12-21:
Dear All,
I have a question about streaming a video to the Icub simulator screen. 
I have successfully setup the ICub Simulator and I can stream the example in READ ME
yarpdev --device test_grabber --name /test/video --mode ball
yarp connect /test/video /icubSim/texture/screen
This works perfectly!. 
I would like to so the same from my C++ code (example project an image to the screen)
I could not find any help on the web for an example a sample code,
I tried to setup a test_grabber device like following 
        PolyDriver texture_send;
	Property p;
	p.put("device", "test_grabber");
	p.put("local", "/test/client"); 
	p.put("remote", "/icubSim/texture/screen");
	texture_send.open(p);
	if (!texture_send.isValid()) {
		printf("Device not available.  Here are the known devices:\n");
        printf("%s", Drivers::factory().toString().c_str());
        exit(0);
       }
       %not sure...
The device can be setup but could not find an interface.  please let me know if you have some suggestions. 
thanks in advance.
Cheers.

pattacini, 2016-12-21:
Dear @robotology/ifi-padawans,

Can you please help @LakshithaPW to solve this problem?

Nicogene, 2016-12-21:
Hi @LakshithaPW,

@giuliavezzani and me found some problems in your code, but we didn't test it, so after the changes we suggest you, if you have other problems let us know :smiley:
You have to put `mode` inside the `Property`, in order to run it in the same  way : `p.put("mode","ball")`.
Another problem is that you are adding the ports' names in fields `local` and `remote` that are not used by the `test_grabber` device. Our advice is to delete the two lines about remote and local and write `p.put("name","test/video")` and then `yarp connect /test/video /icubSim/texture/screen`.

@pattacini please correct us if we're wrong :sweat_smile:

Cheers
Nicolo'

LakshithaPW, 2016-12-21:
@Nicogene 
Thank you very much for the response,
I changed the code to,

```
PolyDriver texture_send;
Property p;
p.put("device","test_grabber");
p.put("name","test/video");
p.put("mode","ball");
texture_send.open(p);
if (!texture_send.isValid()) {
	printf("Device not available.  Here are the known devices:\n");
    printf("%s", Drivers::factory().toString().c_str());
    exit(0);
}
bool ok=Network::connect("/test/video","/icubSim/texture/screen");
if (!ok) {
    printf("Not Successful\n");
    exit(0);
}
```
There seems to be a small issue with the connection `Network::connect`. But it works well in the command prompt. In the connection step it seems nothing is happening. 
Please let me know if you notice any problem with this procedure.
Thanks in advance :)

Nicogene, 2016-12-21:
You have to declare a  Network object, but the good convention is to not do connection in the code

pattacini, 2016-12-21:
To elaborate a bit more, @LakshithaHKUST you ought to:

```cpp
int main() {
    yarp::os::Network yarp;
    if (!yarp.checkNetwork()) {
        yarp::os::yError() << "YARP is not available!";
        return 1;
    }
    // the snippet above kicks in here
    // ...

    return 0;
}
```

However, @Nicogene is right: it's way preferable to establish connections outside code, since we adhere to [**publish-subscribe pattern**][1], where modules should remain as much agnostic as possible of the sources/sinks they receive/send data from/to.

[1] https://en.wikipedia.org/wiki/Publish%E2%80%93subscribe_pattern

Nicogene, 2016-12-21:
Sorry but I commented from the smartphone 😅

pattacini, 2016-12-21:
No problem @Nicogene 🎖

LakshithaPW, 2016-12-21:
Thanks @Nicogene and @pattacini I'll have a look.

pattacini, 2016-12-21:
Closing for the time being then.

closed 2020-12-09 by pattacini