robotology/community #177: Can't start PC104 from Debian 6.0-16.04.05-rt-amd64 image

Tobias-Fischer, 2017-01-11:
Hi @mbrunettini,
I want to update our PC104 to work with the new Debian 6 image. I followed the instructions at http://wiki.icub.org/wiki/The_Linux_on_the_pc104#Burn_the_Debian_image_version_6.x_.28latest.29. Following these instructions, I cannot reach PC104. I've tried both, making the image bootable using the Linux script and Windows script and neither method worked.

By not working, I mean that I cannot ping PC104 from a machine which is connected to PC104 via a switch. The machine's IP is set to 10.0.0.1, subnet 255.255.255.0. A `ping 10.0.0.2` fails.

A few observations:
1) Using the new image, it is really hard to change the network configuration, as the file system is not accessible easily anymore.
2) Trying to boot from the prepared USB on a normal machine fails. It doesn't even get to the stage where it boots from the USB, it seems like as if there is no bootloader installed
3) Unfortunately we don't have a display cable for the PC104, so it's hard to debug :(

It would be great if you could assist.

Thanks a lot and best wishes,
Tobi

Tobias-Fischer, 2017-01-16:
Any news @mbrunettini? @julijenv would it be possible to send one of the VGA cables for easier debugging?

Thanks!

drdanz, 2017-01-17:
@pattacini Perhaps this should be moved to icub-support?

pattacini, 2017-01-17:
Maybe @Tobias-Fischer could open another issue in `icub-support` for the VGA cable, but I'd rather keep this here for what concerns the problems with the image.

mbrunettini, 2017-01-18:
The script used to make the USB bootable is a standard widely adopted by distrubuitions like Filezilla and usually works smoothly **if you follow the instructions carefully**
I suggest to : 
1.  try again the process, reading and following all the instructions (specially: the USB requred min size, the filesystem to format to and how to execute the makebooot script.
2. if the above still fails, try using a different USB pendrive.

pattacini, 2017-01-27:
@Tobias-Fischer is it going any better here?

Tobias-Fischer, 2017-01-31:
I won't get to test this within the next few weeks. I'll close and re-open if necessary.

Thanks, Tobi

closed 2020-12-15 by pattacini