robotology/community #178: Porting Laser Visualiaztion to a Yarp Port

amaroyo, 2017-01-16:
Dear @randaz81, @Tobias-Fischer, @reafrancesco, 

I would like to get some advice for the following problem: I have the iKart in a fixed position (immobile during the whole time) in a room; using the laser I am able to visualize the room and the moving objects using rviz, as you can see in the following pictures: 

[image]
[image]

It is easy to spot the walls, and the green circles are the legs of a person. I would like to know where the person is going in the room, with which velocity and direction. @randaz81 suggested me to use a time difference substraction, so I can distinguish from the walls and the moving objects, i.e. the person.

Together with @reafrancesco, we are planning to use image processing in order to get a cluster of the legs and detect it in the image. So, my real question: 

      **Is there a way to see the visualization of the laser as we see it in rviz, in a yarp port?** 

So then we can easily begin with the image processing. I guess rviz is using some API to get the data from the laser of the iKart, so maybe it's easy to direct that data to a yarp port?

And, is this way too complicated, is there a simpler way?

Thanks a lot!
Alex.

Tobias-Fischer, 2017-01-16:
Hi Alex,
I am not sure whether you should go for image processing. I'd probably directly process the readings from the iKart. Especially as it's not moving, it should be quite easy. There is already a yarp port which outputs these data, which are distances for the various angles.
In the easiest way, you could just store all the distances measured, and then compare them over time to see where objects / humans move. You probably have to deal with some noise etc. but this could be your first step.
It seems like converting it to an image as above introduces an extra step which is unnecessary.

Best,
Tobi

amaroyo, 2017-01-18:
Hi Tobias, 

Yes, you're right. However, if you process it that way, you lose the "meaning" of the data. In the sense, the room and the person moving in the room. If it's getting closer to one part, to the table, to a chair, etc. 
Also, it would be difficult to understand if the readings are correct. If you can visualized it, you can easily check whether everything works fine.

Thanks,
 Alex.

randaz81, 2017-01-19:
You can check the drawing function: `void drawLaser()` inside `yarplaserscannergui`. It uses some basic math and opencv drawing functions to produce an image very similar to the one you posted here. However, that module is a stand-alone application. If you want to manipulate the image data coming from a yarp port, you need to develop your own module that basically takes the image produced inside yarplaserscannergui and pushish it on a yarp port. It should be pretty straightforward..

randaz81, 2017-01-20:
PS: Even the math part in yarplaserscannergui is a good example, I do not recommended to use openCV drawing functions, it's a tedious dependency. Instead, you can use **yarp drawing functions**_http://www.yarp.it/namespaceyarp_1_1sig_1_1draw.html, which are simple to use and do not require external dependencies.

closed 2020-12-08 by pattacini