robotology/community #179: Failing to start an application on MacOS

amaroyo, 2017-01-23:
Hello, 

I am tryting to run an app that works on both, Windows and on Linux but it fails on MacOS. I have installed Yarp and iCub from sources and they both work, as well as other applications. The problem is that once I run the application and I try to read from a port, the following happens:

```
[INFO]Initialization of the processing thread correctly ended
yarp: Sending output from /kInterface/keypressed:o to /tmp/port/1 using tcp
2017-01-23 14:16:27.464 keyboardInterface[44689:916873] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff8209d0db __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff96d2fa2a objc_exception_throw + 48
	2   AppKit                              0x00007fff802bf8b1 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 4471
	3   libopencv_highgui.2.4.dylib         0x000000010d7a40b6 cvWaitKey + 402
	4   keyboardInterface                   0x000000010cd67f81 _ZN16kInterfaceThread3runEv + 417
	5   libYARP_OS.1.dylib                  0x000000010de14a58 _Z18theExecutiveBranchPv + 248
	6   libACE.dylib                        0x000000010e393c3a _ZN21ACE_OS_Thread_Adapter6invokeEv + 254
	7   libsystem_pthread.dylib             0x00007fff97828aab _pthread_body + 180
	8   libsystem_pthread.dylib             0x00007fff978289f7 _pthread_body + 0
	9   libsystem_pthread.dylib             0x00007fff978281fd thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

I have the feeling that the reason behind it, is OpenCV. When I connect to the port to read, the program should open a small OpenCV window but it crashes instead showing the result above. I tried as well to update OpenCV but the problem remains the same. 

I've seen this issue in OpenCV https://github.com/opencv/opencv/issues/7791 

As the application has the typical structure "main, module, thread"; moving the imshow and waitkey to the module solves the problem. However, it's a bit ugly this solution. 

Does anyone has the same problem? Is there a better solution?

Thanks, 
Alex.

francesco-romano, 2017-01-23:
In the issue they are suggesting to move the call in the main thread as the error suggests.
When you say "I moved the call to the module" you did not specify from where you moved the code. I imagine the code was in the Thread, right?
If this is the case then I am afraid this is the only thing you can do (a part re-thinking your application)

vtikha, 2017-01-23:
@amaroyo, the window of OpenCV needs to be opened in the main thread. @francesco-romano is right in saying that you would need to re-think the application but why don't you send the image through YARP to a yarpview instead?

amaroyo, 2017-01-23:
@vtikha, It's not an image what I am sending, I am waiting for a keyboard press to detect it and read it. 

@francesco-romano, yes, I had to move it from the Thread to the Module. I don't really see how much room I have for rethinking the application, as I do not have much more choice. It's an app which waits and reads a keypress from a keyboard, and it sends it to a yarp port.

francesco-romano, 2017-01-23:
@amaroyo the problem is that usually events and graphic function calls must be done in the main thread.  So you have no choice here. Put that in the main thread.
If the thread needs to wait for the command, then you have to synchronise thread and main thread, for example (just an example) by using conditional variables

vtikha, 2017-01-23:
@amaroyo ok, I assumed opening a window would be to display something. It all depends on what you are trying to do, or on what is happening inside your `module` and `thread`. In most of the cases with events and signals it is preferable to have the main thread deal with user inputs and notify other threads of any changes or reorganise things a bit.

amaroyo, 2017-01-24:
@francesco-romano, @vtikha, as this is an auxiliary and very simple app, I do not need to do any synchronisation between threads... but I guess that in a more complex application, it will be more difficult to solve. 

Anyway, OpenCV project have assigned this issue as a bug, maybe in the new release they will have fixed it!

Thanks a lot!
Alex.

closed 2020-12-15 by pattacini