robotology/community #183: CAN RX OVERRUN message on yarprobotinterface

frank-foerster, 2017-02-14:
Hi,

since having figured out how to run and regularly running yarplogger, I noticed that the yarprobotinterface regularly prints error messages such as

`sharedcan[0] board 1 CAN RX OVERRUN`

I'm currently running high-level code issuing commands to the the cartesian interface and ARE in an interleaved manner and did not notice any decrease or breakdown in the functionality of these modules in terms of how the iCub behaves. The error messages do only occur though once I run 
my code so it's definitively cause by moving limbs etc and do not occure constantly but every once in a while. On some runs they don't occur at all.

So I was wondering what this message precisely means and how urgently I should investigate this given the otherwise unimpaired behaviour of the robot.

Many thanks

pattacini, 2017-02-15:
Can anyone of @robotology/firmware-developers take this question?

@randaz81 @marcoaccame ?

pattacini, 2017-02-18:
Hi @frank-foerster 

Let me try to give you a quick feedback on this.

To my knowledge, the warning should mean that there's an amount of time required to process incoming CAN messages on the `board 1` that exceeds the intended limit (i.e. the `overrun`). This is probably due to the position-direct set-points sent by high-level controllers.

Certainly, we shouldn't have the warning and the time slot should be designed more accurately; however, in my experience, I've never spotted any particular impact caused by this overrun to robot behaviors.

randaz81, 2017-02-20:
Actually this warning does not concerns time but the amount of received messages. Each board has a finite (but pretty large) buffer, so one possibility is that the user module is flooding the system with a large amount of commands, thus the boards is discarding some of them.

pattacini, 2017-02-20:
Thanks @randaz81 for the clarification!

High-level Cartesian controllers usually run @ `100 Hz`, meaning that they send one message containing position-direct set-point each `10 ms`. I don't know if this information might help.

julijenv, 2017-02-20:
Hi @frank-foerster ,
How often do you get the error message?
Can you attach a logfile for a normal use of the robot, thx in advance

frank-foerster, 2017-02-21:
Hi @pattacini and @randaz81 ,
many thanks for the explanation. I'm not knowingly generating any messages in my modules which just makes use of the cartesian controller and ARE. But anyways, I assume for the moment that it's ok to run the system given the general absence of funny behaviour.

@julienv: 
I'm getting these kind of messages pretty regularly. I would estimate at least every 2nd run.
I haven't saved any recent logs, but will do so when retesting my software today and post it or link to it here.

julienV, 2017-02-21:
On 21/02/2017 11:52, Frank Förster
      wrote:
    
    
      Hi @pattacini
        and @randaz81
        ,
        many thanks for the explanation. I'm not knowingly generating
        any messages in my modules which just makes use of the cartesian
        controller and ARE. But anyways, I assume for the moment that
        it's ok to run the system given the general absence of funny
        behaviour.
      @julienv:
        I'm getting these kind of messages pretty regularly. I would
        estimate at least every 2nd run.
        I haven't saved any recent logs, but will do so when retesting
        my software today and post it or link to it here.
      —
          it on GitHub, or mute
          the thread.
      
        
          
          
        
        
      
    
    wrong julien !

frank-foerster, 2017-02-21:
Oh dear, sorry for that. I meant @julijenv

frank-foerster, 2017-02-23:
Hi @julienv,

so here is finally one of yesterday's logs containing the relevant error message (line 696)
Before that there are some other error messages of the type

`positionMoveRaw: skipping command because    joint  x is not in VOCAB_CM_POSITION mode`

which probably has to do with me using cartesian control and ARE simultaneously and me maybe not saving and restoring contexts sufficiently often. But I would be surprised if this had anything to do with the "message overflow".

Anyways, here's the log-file:
[yarprobotinterface_220217_1.txt](https://github.com/robotology/QA/files/796216/yarprobotinterface_220217_1.txt)

pattacini, 2017-02-23:
Hi @frank-foerster,

Thanks for the log!

`ARE` make uses of the `Cartesian Controller` underneath, which is in turn responsible for switching the mode in `VOCAB_CM_POSITION_DIRECT` mode. It never switches the joints back to `VOCAB_CM_POSITION`  and never controls the joints in such a modality. It appears from the log that somehow you tried once to control the joints of the arm in `VOCAB_CM_POSITION` (e.g. by means of the GUI while homing?).

Anyway, as you've suspected, the "message overrun" is unrelated to this particular warning.

frank-foerster, 2017-02-23:
Hi @pattacini ,
thanks for the additional information. My application is actually slightly more complicated than I described. I'm using cartesian control, position control, and ARE. Position control normally only in the beginning and end in order to have very tight control of the arm and avoid collision with the table. But I may have slipped some position control command in between. So it's possible that that causes these errors.

pattacini, 2017-02-23:
Ok, then, understood. Remember to change the mode to `VOCAB_CM_POSITION` right before commanding in position.

pattacini, 2017-02-24:
@julijenv there are two overrun events in the log. Do you have any guess on them?

julijenv, 2017-02-24:
well, I think it is exagerated to think that this could affect in any kind of way the activity of the robot if it was more often or in any specific movement so I would be worried but so far I would leave it like that.

pattacini, 2017-02-25:
Yes, definitely I'm not worried about, but you know, it's always better to keep every warning under control 😉 

@frank-foerster shall we close this?

frank-foerster, 2017-02-25:
Hi @julijenv , thanks for the message of appeasement. That's what I wanted to hear.
Just wanted to be sure that nothing serious is amiss and I should look into that before cotinuing to use the iCub.
@pattacini : yes, closing this for now.

closed 2020-12-15 by pattacini