robotology/community #184: Graphics libraries and HardDisk based PC104

mbrunettini, 2017-02-14:
On the new PC104 with hard disk, graphics-based tools have not been installed (except the ones required for firmware interfaces) and it is better not to install them. Anyway, to use graphic tools (like editors, files managers, etc..) is very useful and sometimes essential.
Instead of running remotely a tool and displaying it locally (like we did), we can **mount locally via SSH the PC104 filesystem** and run locally all the tools we need.
In order to mount locally the PC104 filesystem : 
1) install on your client the pacakge sshfs
`sudo apt-get install sshfs`
2) create a mount point for the remote pc104 filesystem
(eg.) `mkdir /home/icub/pc104-fs` 
3) mount the remote /usr/local/src/robot via SSH
`sudo sshfs -o allow_other icub@10.0.0.2:/usr/local/src/robot/ /home/icub/pc104-fs`

ifyou already setup the passwordless SSH login, you can skip the password:
`sudo sshfs -o allow_other,IdentityFile=/home/icub/.ssh/id_rsa  icub@10.0.0.2:/usr/local/src/robot/ /home/icub/pc104-fs`

mbrunettini, 2017-02-14:
I will add a command alias on the client machine environment, so the process will be simpler.

traversaro, 2017-02-15:
cc @robotology/icub-developers

S-Dafarra, 2017-05-29:
Thanks for the guide! I was wondering, what is the correct procedure to unmount the remote from the client once I have done working on it?

francesco-romano, 2017-05-29:
@mbrunettini what about adding a line in `/etc/fstab` for that mount point?
Something like:

```
sshfs#icub@10.0.0.2:/usr/local/src/robot/  /mnt/icub-head   fuse    defaults,allow_other    0  0
```
In this way, mounting and unmounting is performed through the `mount` utility.

Than, you are more expert than me regarding the proper options to put in fstab, like `noauto` or `user`, etc...

mbrunettini, 2017-05-30:
The unmount procedure is the same as a normal mounted filesystem:
`sudo umount _MOUNT_POINT_`
or 
`fusermount -u _MOUNT_POINT_`

closed 2020-12-12 by pattacini