robotology/community #185: Jacobian in the ICub Simulator

pattacini, 2017-02-15:
@LakshithaPW commented on [Wed Feb 15 2017](https://github.com/robotology/icub-main/issues/423)

Dear All,

I was wondering anyone could help me to find the Jacobian in real time for the iCub Simulator ?
I want to control the end effector of the iCub given a specified task space velocity. 

Thanks in advance

pattacini, 2017-02-15:
Hi @LakshithaHKUST 

I've just moved this issue to our [**Q&A**][1] system, where you'll receive a reply quite soon 😉 

[1]: https://github.com/robotology/QA

pattacini, 2017-02-16:
Hi @LakshithaPW 

To get the **Jacobian** of the arm (or any other limb) of the iCub (no matter if we consider the simulated or the real robot), you might want to use [`iKin` library][1], whose tutorials are available from our [manual][2].

Essentially, you ought to use a snippet looking pretty much the following:
```cpp
#include <iCub/iKin/iKinFwd.h>

iCub::iKin::iCubArm arm("left");   // declare the arm object
arm.setAllConstraints(false);   // we don't need to update the limits from the robot to compute the Jacobian
arm.releaseLink(0);   // release torso pitch joint
arm.releaseLink(1);   // release torso roll joint
arm.releaseLink(2);   // release torso yaw joint

// retrieve here joints from the torso and the left_arm
yarp::sig::Vector torso(3), left_arm(16);
ipos_torso->getEncoders(torso.data());
ipos_left_arm->getEncoders(left_arm.data());

// fill in the vector of degrees of freedom
yarp::sig::Vector dofs(3+7);
dofs[0]=torso[2];   // be careful of the order here
dofs[1]=torso[1];
dofs[2]=torso[0];
dofs[3]=left_arm[0];
// ...
dofs[9]=left_arm[6];

yarp::sig::Matrix Jacobian=arm.GeoJacobian((M_PI/180.0)*dofs);   // iKin works with radians
```

Besides recruiting directly the Jacobian, you might also consider relying on the [Task-Space Reference Velocity Command][3] of the Cartesian Interface.

[1]: http://wiki.icub.org/iCub/main/dox/html/group__iKin.html
[2]: http://wiki.icub.org/iCub/main/dox/html/icub_tutorials.html 
[3]: http://wiki.icub.org/iCub/main/dox/html/icub_cartesian_interface.html#sec_cart_taskrefvel

LakshithaPW, 2017-02-16:
Thank you very much for the response..
I'll follow the snippet.

pattacini, 2017-03-30:
Related issue #208.

closed 2020-12-08 by pattacini