robotology/community #186: icub joint specification

nedatym, 2017-02-17:
Hi,

In order to solve an optimization problem, I should have maximum feasible torque which can be applied on each joint. Also I need maximum admissible velocity of each joint and moments of inertia for all icub links.
It's appreciated if you could provide this information. 
Thank you.

Best, 
Neda

pattacini, 2017-02-17:
@traversaro @DanielePucci @francesco-romano 
do you have such information?

DanielePucci, 2017-02-21:
Hi @nedatym,
if you could provide us with additional information about the optimisation problem, we may have additional insights into the big picture and give you more precise suggestions.

In general, during torque-controlled experiments, we set the maximum (desired) joint torque differently for each joint. As a rule of thumb, you can set (unit measurement is Nm):

`arms-max-torque`: 12 
`torso-max-torque`: 24
`legs-max-torque`: 40

The `legs-max-torque` is pretty high because during highly-dynamic-balancing [[1]], we may reach high torque values. For other applications, though, you can bring it down to lower values without problems. 

Concerning velocity, for the time being we do not set any maximum value for the (desired) joint velocity. As rule of thumb, I would keep them below `45 [deg/s]`, but this number depends on the application and @pattacini may suggest better values.

Finally, it is not really clear to me what you mean by: 

`I need maximum admissible velocity [...] of inertia for all icub links.`

The link inertia values are constant numbers (use them carefully because they depend upon the frame w.r.t. which to express the inertia), and can be found in the icus models, e.g.

https://github.com/robotology/icub-gazebo/blob/master/icub/icub.sdf#L10

[1]: http://ieeexplore.ieee.org/document/7803266/

randaz81, 2017-02-21:
If you need to abstract your application from a specific robot, `yarp::dev::ITorqueControl::getTorqueRanges() ` should return the information you need.
In the case of iCub, the function mentioned above returns the values stored in the iCub configuration files, which are the numbers mentioned by Daniele in the comment above.

pattacini, 2017-02-25:
@nedatym any feedback on this?

pattacini, 2017-02-28:
Closing for the time being.
Feel free to open it again if required.

closed 2020-12-15 by pattacini