robotology/community #188: Port callback threading

billingo, 2017-02-20:
I have a Yarp module listening to input messages using BufferedPort::onRead(...). I noticed that these callbacks are blocking, meaning that the call to onRead() must finish before the same function can be called again, in response to the next message. In my implementation, the processing does at times take too long, meaning that messages will get lost since the single-threaded calls to onRead are not able to keep up with the phase of incoming messages. 

I would prefer onRead to be called in parallell, for example using an underlying thread pool. I realize I could of course implement the threadpool myself, and let each call to onRead distribute new tasks to the pool, but before I go ahead with this I just wanted to ask if anyone on the list have a better suggestion, maybe by the use of a different reader. 

Kind regards, 
-- 
Erik Billing,
Assoc. Senior Lecturer,  
University of Skövde
www.his.se/erikb
erik.billing@his.se

pattacini, 2017-02-21:
Hi @billingo 

Although I'm in favor of your solution using thread pooling (it's cleaner and safer), have you ever tried to use [**`yarp::os::PortReader`**][1] instead? Perhaps, I'm not sure, it allows for **reentrant callbacks**. Here's a [**tutorial**][2] (you don't need to provide a reply in your case).

Anyone has more heads-up on this? @traversaro @drdanz 

[1]: http://www.yarp.it/classyarp_1_1os_1_1PortReader.html
[2]: https://github.com/robotology/yarp/blob/master/example/port_power/ex0300_port_callback.cpp

lornat75, 2017-02-24:
I agree, switching to Portreader will allow concurrent callbacks.

billingo, 2017-02-25:
Perfect, many thanx. I'll try this out!

closed 2020-12-15 by pattacini