robotology/community #189: iCub Basic Image Processing question

vtikha, 2017-02-25:
_From @accateo on February 25, 2017 10:37_

Hi,
I'm trying to follow the "Basic Image Processing" tutorial of iCub:

http://wiki.icub.org/brain/icub_basic_image_processing.html

I have connected the image source, but I don't understand where I should see these messages:

We got an image of size 320x240
  We got an image of size 320x240
  We got an image of size 320x240
  ...
in what terminal? 
Second question: why in the iCub_Sim terminal I have always this warning?

`[WARNING]the simulation is too slow to run in real-time, you should increase the timestep in ode_params.ini (current value: xx, suggested value: xx)`

I have also changed the timestep value in ode_params.ini file, but the warning remains and change its value.

Thanks

_Copied from original issue: robotology/icub-support#410_

vtikha, 2017-02-25:
Hi @accateo, first of all, please use the [Q&A](https://github.com/robotology/QA/issues) for any questions you have and not the icub-support which is a repository used to signal failures/repairs or ask for hardware/software questions regarding the physical robots. I will close this and reopen it in the proper repo.

vtikha, 2017-02-25:
@accateo 

1. tutorial basically generate two different programs. One called `findLocation` and one `lookAtLocation`. The first one, `findLocation`  basically generates the prints you are looking for: 
> We got an image of size 320x240

If you have copied the code from the online tutorial, then these prints happen in the terminal where you run the `findLocation` module. 
If on the other hand you are using the `icub-tutorial` repository, just uncomment [this line](https://github.com/robotology/icub-tutorials/blob/master/src/imageProcessing/findLocation.cpp#L37) from the `findLocation.cpp` and recompile. This will now display the printouts at every frame.

The second one, `lookAtLocation` program basically moves the head towards the bluish object located on the table.

2. The warning you see in the `iCub_Sim` is just telling you that your system is not able to make it run in real time. 
This is not tragical at your stage, but I am guessing that you are using all the parts of the `iCub_Sim`. Try removing some parts in the `iCub_parts_activation.ini` file. 
Depending on the installation you followed, locate this file and set the parts that you do not need to `off` eg: `left_hand off` and `right_hand off` etc ... 
In order to easily locate this file run the following command in a terminal: ``yarp resource --context simConfig --from iCub_parts_activation.ini`` 
This will provide the full path to the file that is loaded by the simulator at runtime, therefore just edit that one. Since you are editing the file, make sure that `objects` are set to `on` to load the table and blue ball that are needed for the tutorial.  
Please find some further documentation on the simulator [here](http://wiki.icub.org/wiki/Simulator_README)

accateo, 2017-02-25:
Thanks for the second question.
For the first question instead, what do you mean with "these prints happen in the terminal where you run the findLocation module.". I have different terminal: yarp server, icub sim, icub view.
I connect the image using the command:

yarp connect /icubSim/cam/left /tutorial/image/in
Should I open other terminals?

vtikha, 2017-02-25:
@accateo where do you run the executable findLocation? How does the `/tutorial/image/in` even get created? is that the port name you are giving to the `yarpview`? Then you are seeing images from the `simulator's camera` on the `yarpview` correct?

vtikha, 2017-02-25:
@accateo what to run in terminals (I am assuming you are following the tutorial and compiled code)
```
- yarpserver
- iCub_SIM
- findLocation 
- yarp connect /icubSim/cam/left /tutorial/image/in
```

accateo, 2017-02-26:
Sorry but "findLocation" terminal how can I open it?
I have already generate yarpy executable file...

pattacini, 2017-02-26:
Hi @accateo 

What @vtikha pointed out is simply that those printout messages are showing up in the terminal from which you run the `findLocation` executable.

Can you launch the tutorial again to verify that?

vtikha, 2017-02-26:
Hi @accateo ok now I get it, so, you created a `findLocation.cpp` file, copied the example code from the tutorial and did 
```
yarp cmake
```

Then you compiled and this  generated a `yarpy` executable. What you did not do is rename the file as mentioned by replacing the project name `yarpy` in following line of the `CMakeLists.txt` with `findLocation` or whatever you would like to.
```
set(KEYWORD "yarpy")
```

So, no worries, now just run the `yarpy` executable (or whatever you chose if you changed the project name as mentioned above) in a terminal instead of `findLocation` as previously mentioned:

```
- yarpserver
- iCub_SIM
- yarpy 
- yarp connect /icubSim/cam/left /tutorial/image/in
```

You should now be ok

accateo, 2017-02-27:
The problem is that I have exactly done the things that you have described, but these messages don't appear. Anyway is not important, the rest of the simulation works.
Thank you

vtikha, 2017-02-27:
@accateo can you post the code here please? I am referring to the code that is used to generate the `yarpy` executable. If you followed the tutorial it should be called `find_location.cpp`

accateo, 2017-02-28:
#include <stdio.h>
     #include <yarp/os/all.h>
     #include <yarp/sig/all.h>
     using namespace yarp::os;
     using namespace yarp::sig;
     int main() {
      Network yarp; // set up yarp
      BufferedPort<ImageOf<PixelRgb> > imagePort;  // make a port for reading images
       imagePort.open("/tutorial/image/in");  // give the port a name
       while (1) { 
           ImageOf<PixelRgb> *image = imagePort.read();  // read an image
           if (image!=NULL) { // check we actually got something
               printf("We got an image of size %dx%d\n", image->width(), image->height());
           }
       }
     return 0;
     }

vtikha, 2017-02-28:
@accateo you are probably not running the code or not connecting as everything is in order. 
If there is an image coming in it will print the message. Could you perform a small test?  
After running the `yarpserver` and the `iCub_SIM`, could you launch the following commands in a terminal?

```c++
yarpview --name /viewLeft 
```
and in another one do

```c++
yarp connect /icubSim/cam/left /view/Left
```

Can you confirm that you see an image displayed in the yarpviewer?

vtikha, 2017-02-28:
@accateo then it is impossible that you do not see the printouts unless the connections you are doing and executable that you are launching are not correct. 
I am referring to the excerpt of the above snippet:

```cpp
ImageOf<PixelRgb> *image = imagePort.read();  // read an image
if (image!=NULL) { // check we actually got something
    printf("We got an image of size %dx%d\n", image->width(), image->height());
}
```

accateo, 2017-02-28:
I know that is really strange, i don't know. But I repeat, it is not important.
Thanks for the patience

pattacini, 2017-02-28:
Ok then, I think we can close this issue.

closed 2020-12-09 by pattacini