robotology/community #190: CAN FW update questions.

pedrombmachado, 2017-02-27:
Dear all, 
  I am trying to upgrade the CAN FW update on iCub 2.5 (Eth base). I am able to connect to see CAN 1 and CAN 2. However I am not sure where to find the FW for each one one of CAN 1 and 2. I have updated CAN1 with the EMS.hex. and I got the following:
[DEBUG]  BUS:id   type     version
[DEBUG] CAN1:4       8     2.1.-1
[DEBUG] CAN1:1       8     2.1.-1
[DEBUG] CAN1:2       8     2.1.-1
[DEBUG] CAN1:3       8     2.1.-1
[DEBUG] CAN2:13       6     3.0.4
However the version is lower. I think the type is now correct, but I lost the S/N of strain and it now says PROTOCOL N/A.
Therefore, I may be using the wrong FW version. should I use MCP2PLUS and/or MCP4PLUS? If so what file is to be downloaded to CAN1 and CAN2?
Many thanks in advance.
Pedro

randaz81, 2017-02-27:
@marcoaccame can you help Pedro?
@pedrombmachado  can you please specify the name of your robot?

pattacini, 2017-02-27:
The robot should be **`iCubNottingham01`** 🤖

pedrombmachado, 2017-02-27:
Hi @pattacini ,
   Thanks for adding the robot name :).

marcoaccame, 2017-02-28:
Hi @pedrombmachado , 

as a general rule, instructions on how to update FW on ETH robots are on the icub wiki: http://wiki.icub.org/wiki/Firmware#Updating_the_FW_of_the_ETH_boards.

The link tells how to update the FW of both ETH and CAN boards.

If you read the wiki I am sure most of your doubts will be answered.

About your particular case, I can see that you are receiving support for how to perform FW udate for your robot iCubNottingham01  in another issue: https://github.com/robotology/icub-support/issues/407.

I would recommend to keep any request about you need of FW update in the the above issue.

But if it can help, I can already tell you that no FW update of CAN boards is required on your robot, because iCubNottingham01  was delivered around August 2016 and since then we did not release any fundamental FW for CAN boards.

If by mistake you wrongly mis-programmed some board I am available to help you to restore the status quo ante inside the above icub-support issue.

marcoaccame, 2017-02-28:
Hi @pedrombmachado, I close the issue for now and I will help you for any specific problem you may have in the relevant issue http://wiki.icub.org/wiki/Firmware#Updating_the_FW_of_the_ETH_boards.

closed 2020-12-15 by pattacini