robotology/community #191: Virtual Skin

emilio-cartoni, 2017-02-28:
Dear all,
I have been told that a few years ago a "Virtual Skin" for the iCub was developed that basically "simulated" all movement commands in advance to avoid the iCub hitting itself.
I wanted to try and test it but I only found this page http://wiki.icub.org/wiki/MoBeE_installation which seems to be outdated (the source apparently fails to compile because something in Yarp has changed meanwhile).

So my question is: is the VirtualSkin still supported?
More generally how would you ensure that all movements from iCub are "safe" (meaning that it doesn't break by hitting itself)?

pattacini, 2017-03-01:
Hi @emilio-cartoni 

I believe you're referring to our implementation of [**Peripersonal Space**][1] (PPS) on the iCub with the term "_virtual skin_". Actually, PPS does not account for a virtual skin but rather for a representation of the space surrounding the robot body that can be used in turn to predict contacts by merging **visual and tactile information**.

That being said, PPS equips the robot with a further **perceptive layer**, therefore you need to rely on a robust and reliable visual tracker to elicit "virtual" activation of the skin parts that we predict will be in touch with the approaching object (posing a threat to the safety of robot movements).

The key aspects here are thus: _robust perception_, _reactive control_.

By contrast, if you aim to achieve safe movements - as for the main point you made - without the need for complicated trackers and other perceptive magic, I'd recommend you to rely on **planning components**, among which **MoBe** is clearly a valid example, although we don't maintain it (you could reach out to the authors directly).

Alternatively, you might want to use our [**Cartesian Control**][2] system that is responsible for planning and performing movements of the iCub end-effectors (mostly hands) in the operational space. In particular, this component provides movements of the iCub arms that are guaranteed not to hit the torso with the upper arm covers as well as it takes into account mechanical limitations imposed by the length of the shoulder's cables.

What is left to the user is to check whether the hand will collide against the iCub body (torso + other hand). However, these checks under given conditions are fairly manageable boiling down to simple geometric verifications: e.g. the Cartesian location of the hand under use must be outside a cylinder centered around the torso.

Of course, we could also extend the inverse kinematics problem to account for such collisions at an earlier stage at the cost of degraded tracking performance; nonetheless, in our experience, this level of sophistication is not really required for a wide variety of Cartesian tasks.

Don't hesitate to ask for further details if you have other questions and/or doubts on the implementation.

[1]: https://github.com/robotology/peripersonal-space
[2]: http://wiki.icub.org/iCub/main/dox/html/icub_cartesian_interface.html

emilio-cartoni, 2017-03-02:
Hi @pattacini,
thanks, we were thinking more along the planning components approach but PPS has some interesting features we might need, I'll look into the papers.

Our situation is that we have some projects starting that will all involve the iCub exploring a table with a few objects on it and we would like to have a general way to ensure that the iCub can freely explore any kind of movements but without the risk of breaking the iCub by hitting itself or other objects.

From what we have gathered so far, we could do the following:
1) Set iCub interaction to compliant mode
2) Enforce that all movements will be slow speed
3) Use Cartesian control to avoid hitting itself and the table (geometric checks against a cylinder for torso and a plane for the table)

We would still hit the objects but we do have to interact with them anyway and it shouldn't be a problem given that the collision will be slow speed (I hope!).

On this aspect, it might  be nice to have something like PPS that tracks (multiple) objects on the table and adjusts the speed when near them (so that we could go fast when in "free air" and automatically slow down near the objects).

Some of the projects would like to control the iCub in the joint space, so we would need a translation from the joint space to the Cartesian space (using iKin forward kinematics I guess) before sending their commands through the Cartesian controller. 
(I have a feeling this translation could result in some weird behaviour and pose some challenges to the algorithms trying to control and learn in the joints space).

All of this should be as transparent as possible for each project.
So, from a software point of view, we would probably create some kind of proxy where the controller of each project does not interact directly with the iCub but sends the commands first to the "safety proxy" which does the checks and then executes the command (I think MoBe used this approach too).
The controllers will probably stream the desired positions so the "safety proxy" should be fast in doing the translation/checks.

Does it look like a reasonable approach?

pattacini, 2017-03-02:
That looks all good to me.

Let me only clarify one aspect: PPS does not track objects; it's merely responsible for eliciting virtual activation of the skin taxels involved in the interaction, whose inputs to do so are provided externally by any kind of tracking mechanism.

For tracking objects in a table-top scenario, I'd suggest you give a glance at our [**`IOL`**][1] component instead. There's an explanatory video showing `IOL` in action here: https://youtu.be/5PfqoNpPbAE.

Regarding `IOL`, we don't have a fully documented API yet, and it might be likely that we would need to iterate together to check the installation and to introduce you to its basic functionalities. However, it's doable, since we did it already in the past with several other partners, who can even chime in with their contributions too.

>I have a feeling this translation could result in some weird behaviour and pose some challenges to the algorithms trying to control and learn in the joints space.

It's definitely true 😉 

[1]: https://github.com/robotology/iol

pattacini, 2017-03-07:
Closing for the time being. Feel free to open it again when needed.

emilio-cartoni, 2017-03-08:
> That looks all good to me.
> Closing for the time being. Feel free to open it again when needed.

@pattacini Ok, thanks for the help! 
We'll have a look at IOL too.

closed 2020-12-08 by pattacini