robotology/community #192: Docker integration

diegoferigo, 2017-03-03:
Dear all,

I recently started playing with [docker](https://www.docker.com/) containers, and I realized that our infrastructure does not exploit at all this tool that [can be very helpful](https://www.docker.com/what-docker). For who doesn't know, docker allows the creation of the so-called *containers*, that are basically isolated root trees of a Linux OS. Docker containers could be seen as light VMs, where the entire hardware virtualization is substituted by the sharing of the host's kernel performed by a specific [additional layer](https://www.docker.com/what-container).

I won't discuss more details here. Long story short, I tried to integrate docker into Travis CI (robotology/gazebo-yarp-plugins#286) in order to enhance the testing capabilities and avoid manual edits that as far as I understood are required before the merger of different branches. I'd like to start a discussion on a possible docker integration into our infrastructure also for other tasks. I'll describe the basic elements of the system I built so far.

The images I used are stored into [robotology-playgrounds/docker_images](https://github.com/robotology-playground/docker_images) repository. The images have been uploaded to a [DockerHub account](https://hub.docker.com/r/robotology/), platform that hosts the Dockerfiles and automatically builds the images. Whenever *docker_images* receives new commits, a new generation is triggered.

The images I created are the following:
1. **yarp**: installation from binaries (*yarp:bin*), from git master branch (*yarp:master*), from git devel branch (*yarp:devel*)
2. **yarp-gazebo**: image 1. with gazebo 7 and 8 (all combination: *yarp-gazebo:gazebo7bin*, *yarp-gazebo:gazebo8bin*, *yarp-gazebo:gazebo7master*, ..., *yarp-gazebo:gazebo8devel*)
3. **yarp-tdd**: image 2. with packages for Test Driven Development built from sources (*yarp-tdd:gazebo7bin*, *yarp-tdd:gazebo8bin*, ..., *yarp-tdd:gazebo8devel*)
4. **icub**: image 1. with icub packages from ppa (I didn't spend much time on this setup)

For features, TODO tasks, and general comments regarding Travis CI integration, please check the pull request robotology/gazebo-yarp-plugins#286. The *docker_images* repository is still in early development stage, and contributions / suggestions are more than welcome. Let me know if you see possible future use cases that require significant changes to the structure I created.

cc @drdanz @traversaro @francesco-romano

pattacini, 2017-03-03:
Interesting discussion @diegoferigo 
@lornat75 may want to participate.

giodegas, 2017-03-04:
please have a look to my two years old docker image I am maintaining every yarp release 
http://hub.docker.com/r/giodegas/yarp

I also encaplusated a full gazebo simulator with a WebGL GUI:
http://hub.docker.com/r/giodegas/gzweb

and experimented to inlude the iCub simulator:
http://hub.docker.com/r/giodegas/icub-gzweb

I am glad that finally official docker releases are born. 
Please fell free to merge my docker files if usefull.

Cheers
Giovanni

pattacini, 2017-03-04:
@giodegas's contribution is detailed in #24

diegoferigo, 2017-03-06:
@giodegas 
I'm aware of your project, those images were the only one I found with YARP, thanks for maintaining them so far. Considering your longer experience with docker, feel free to advance suggestions.
For what concern `gzweb`, I still don't know anyone here at IIT that uses it. I'll merge your docker file if someone will request it.

pattacini, 2017-03-12:
Closing, feel free to open it again when needed.

closed 2020-12-08 by pattacini