robotology/community #193: Issue in downloading the virtual machine

LCaranzi, 2017-03-06:
Dear all,

I had troubles in downloading the VM. I managed to connect to the server (ftp.iit.it), but I do not have access to the subdirectory of the server, thus I cannot download the file iCub_Ubuntu_16.04_x64.zip. The error message is "Failed to retrieve directory listing" as it can be seen in the image below.
Can someone help me?
Many thanks.
[image]

pattacini, 2017-03-06:
Hi @mbrunettini 
Just for your reference, I've tried to download the VM via FileZilla and I was successful.

Could you help @LCaranzi?

mbrunettini, 2017-03-07:
You must use the FTPS protocol (port 990), it seems that you connected using the default FTP port.
Ref [the following instructions](https://github.com/vvv-school/vvv17/wiki/Before%20arriving%20at%20VVV)

LCaranzi, 2017-03-08:
Ok thanks, it worked!

closed 2020-12-15 by pattacini