robotology/community #195: icub-firmware-build

pedrombmachado, 2017-03-09:
Hi all,
   I am trying to fix my environmental variables and I have a question. On the {wiki](http://wiki.icub.org/wiki/ICub_PC104_Linux_Image_-_Detailed_info) it says:
export FIRMWARE_BIN=${ROBOT_CODE}/icub-firmware/build
however the repository name is:
[icub-firmware-build](https://github.com/robotology/icub-firmware-build)
and therefore shouldn't it be...
export FIRMWARE_BIN=${ROBOT_CODE}/icub-firmware-build
Many thanks,
Pedro

pedrombmachado, 2017-03-09:
Found the answer. I was missing the icub-firmware repository.

pedrombmachado, 2017-03-09:
Found the repository however it does not contain the build folder. 
Shall I leave it like this?
export FIRMWARE_BIN=${ROBOT_CODE}/icub-firmware

randaz81, 2017-03-09:
- `icub-firmware` contains the sources of the boards firmware. You don't need to download that repo, it is simply useless unless you have the firmware compiler(s).
- `icub-firmware-build` contains the binaries, generated from `icub-firmware` sources. It's the folder you need to check when you want to upgrade the firmware of robot boards.
 - `icub-firmware-shared` contains some sources and definitions needed to compile some plugins of `icub-main` necessary to run yarprobotinterface on the pc104 (e.g. embObjMotionControl)

- Currently, I'm not aware of any practical use of the environment variable `FIRMWARE_BIN`. I think it has been reserved by developers for future use, so please simply ignore it.

pedrombmachado, 2017-03-09:
Hi @randaz81 
  Many thanks for the reply. If possible please update the [Readme](https://github.com/robotology/icub-firmware-shared) to reflect that.
Many thanks,
Pedro

traversaro, 2017-03-09:
I opened a PR for adding a README to `icub-firmware-build` one year ago explicitly to avoid this confusion: https://github.com/robotology/icub-firmware-build/pull/8 , nobody ever commented, so I think we can merge it. : )

pattacini, 2017-03-09:
Go ahead 😉 then @traversaro

closed 2020-12-15 by pattacini