robotology/community #198: compilation of iCub-main on pc104 - ENABLE_icubmod_icubheadcalibratorV2

pedrombmachado, 2017-03-09:
Dear all,
  I am compiling the icub-main as decribed in the [wiki](http://wiki.icub.org/wiki/Compilation_on_the_pc104).
  I cannot find the option
**_ENABLE_icubmod_icubheadcalibratorV2_**
and I am using the latest version of the [icub-main](https://github.com/robotology/icub-main) with CMAKE 3.0.2.
Is ENABLE_icubmod_icubheadcalibratorV2 obsolete and if so should I enable anything else in its place?
Many thanks,
Pedro

barbalberto, 2017-03-09:
That device does not exists anymore, you can safely ignore it.

pattacini, 2017-03-09:
That's fine, but then we should also update the instructions pointed above.
@randaz81 @barbalberto would you be willing to take on this?

barbalberto, 2017-03-09:
Done

pedrombmachado, 2017-03-09:
Just a quick suggestion. Please make sure that says something like icub ver... **_and above_**. It really helps the user to follow the instructions.
Many thanks.
Pedro

pedrombmachado, 2017-03-09:
Hi @barbalberto ,
 We have the icub v2.5 and there are at least 3 flags missing [icub-support/issues/407](https://github.com/robotology/icub-support/issues/407):
The device embObjStrain is not compiled. It is needed to manage the F/T sensors. See point 1 for how to solve.
    The devices embObjMais and canBusSkin are not compiled.

closed 2020-12-15 by pattacini