robotology/community #199: tutorial_yarp-ports problem

accateo, 2017-03-11:
Hi,
I'm following the tutorial_yarp-ports:

https://github.com/vvv-school/tutorial_yarp-ports

I have compiled the "port folder" code and the "buffered port" code. After that, I have run the two executables and the yarp server works correctly.
I have also connect the ports:

`$ yarp connect /w /relay/in`
`$ yarp connect /relay/out /r `

but when I try to write in the "yarp write" terminal, I don't see the command sent in the "yarp read" terminal.
I have a doubt. I did everything correctly? Or I have to edit one of the two "main.cpp" and add to it a second port?

Thanks

pattacini, 2017-03-12:
Does anyone from @robotology/ifi-younglings want to take on this?

@Iaxama @Arya07 @gmmarconi ?

Arya07, 2017-03-13:
Hi @accateo,
yes, to complete the tutorial, you have to edit the cpp files, following the commented hints.

pattacini, 2017-03-13:
The complete tutorial is contained in the branch [`solution`](https://github.com/vvv-school/tutorial_yarp-ports/tree/solution).

closed 2020-12-15 by pattacini