robotology/community #200: PC104 backup

BrutusTT, 2017-03-13:
Hi,

is there an image for the pc104 or an installation guide or something similar (iCub Edinburgh)? 

I just want to make sure we have a backup copy before changing stuff on the head. In the past there was a usb stick but that seems to be gone now. Any suggestions to get a backup that I can play back easily?

Cheers

pedrombmachado, 2017-03-14:
Hi @mbrunettini ,
   Have you considered to use Deja-Dup? It is a very simple backup tool that works and allows you to save your information. You can also use dd if you want to clone the all image:
run from remote computer

$ dd if=/dev/sda | gzip -1 - | ssh user@local dd of=image.gz

run from local computer

$ ssh user@remote "dd if=/dev/sda | gzip -1 -" | dd of=image.gz

Live example

$ ssh skinner "dd if=/dev/sda5 | gzip -1 -" | dd of=image.gz
208782+0 records in
208782+0 records out
106896384 bytes (107 MB) copied, 22.7608 seconds, 4.7 MB/s
116749+1 records in
116749+1 records out
59775805 bytes (60 MB) copied, 23.9154 s, 2.5 MB/s

$ ll | grep image.gz
-rw-rw-r--.   1 saml saml  59775805 May 31 01:03 image.gz

Methods for monitoring?

    Login via ssh in another terminal and ls -l the file to see what it's size is.

    You can use pv to monitor the progress of a large dd operation, for instance, for the remote example above, you can do:

    $ dd if=/dev/sda | gzip -1 - | pv | ssh user@local dd of=image.gz

    Send a "SIGUSR1" signal to dd and it will print stats. Something like:

    $ pkill -USR1 dd

BrutusTT, 2017-03-14:
Thanks, should have looked at stack overflow earlier ;). 

I thought that there is maybe an image of some sort around for shipped pc104s. But I will try the dd way.

gsaponaro, 2017-03-14:
Hi @BrutusTT,

> I just want to make sure we have a backup copy before changing stuff on the head. In the past there was a usb stick but that seems to be gone now. Any suggestions to get a backup that I can play back easily?

The "[Linux on the pc104](http://wiki.icub.org/wiki/The_Linux_on_the_pc104)" page of the manual points out to how to download and use the PC104 Debian images (http://www.icub.org/download/pc104-images/).

The "[Compilation on the pc104](http://wiki.icub.org/wiki/Compilation_on_the_pc104)" page is also useful, describing the kind of customization usually needed on the PC104: environment variables defined in `.bash_iCubrc`, custom CMake flags for the iCub software. You may want to back up these configurations. Hope this helps.

BrutusTT, 2017-03-14:
Hi @gsaponaro 

Our "problem" is that we got Ubuntu instead of Debian on the pc104. Therefore, I was not quite sure if all of the instructions still apply to our version.

pedrombmachado, 2017-03-15:
Hi @BrutusTT ,
  Great news then.

pattacini, 2017-03-15:
@pedrombmachado @BrutusTT thanks for sharing the instructions.
@mbrunettini might consider putting them in our wiki too, as an alternative method to back up the system.

mbrunettini, 2017-03-17:
As @gsaponaro suggested, in [this page](http://wiki.icub.org/wiki/The_Linux_on_the_pc104) on iCub wiki you can find instructions on how to download and install a pc104 image.
As regarding backup of a system: if you have a PC104 with persistency (v5 and v6) you simly have to get USB memory inside iCub's head, connect it to another computer (better a linux one) and inside the PERSISTENCE partition you can find all the files you have modified. Just save them somewhere. Simply as this.

BrutusTT, 2017-03-17:
Since we have NOT a usb stick (at least as far as I could spot from opening the head), where would I find the drive on our head? We also do NOT have Debian! So sorry to say but its not as simple to me. Downloading the image does not work either as there is no Ubuntu image as far as I have seen.

Anyway, pls keep it closed.

mbrunettini, 2017-03-17:
Which version do you have?

closed 2020-12-08 by pattacini