robotology/community #201: About using the world interface plugin

charleswilmot, 2017-03-14:
Hi there!

My name is Charles Wilmot, I'm a phd student at the _Frankfurt Institute for Advanced Studies_.
Together with labs in Darmstadt (TUDa) and Rome (CNR), we chose Gazebo to simulate the iCub robot. Our purpose is to study intrinsic motivation and self generated goal learning.

We need your help because we're having a deadlock. Concretely, we are using the language Python to control the iCub over YARP. So far, we can make the virtual robot move and we can receive images from the cameras. Our main remaining issue is to add, move and delete virtual objects in the world. I've seen that there is a dedicated plugin for that task (world interface), but it is not trivial to me how plugins are working and how to use them. We would really appreciate if you could give us some hints.

Maybe my misunderstanding comes from the fact that I'm using YARP over the Python bindings, and not from C++.

Is there a code snippet that creates object in Gazebo that I could inspire from?
What is the basic mechanic behind plugins in general?

Thank you very much,
Charles

Related issues: #159 #258 #273
Link to the CORDIS sheet of our project: [GOAL-Robot](http://cordis.europa.eu/project/rcn/203543_en.html)

pattacini, 2017-03-14:
Hi @charleswilmot 

Could you please refine your related issues, since they don't seem to be real links. Follow instructions given in #3 for providing cross-references from different repositories.

cc @traversaro @francesco-romano @randaz81

charleswilmot, 2017-03-15:
Hi @pattacini ,

Sorry for that, I updated my question now.

pattacini, 2017-03-16:
Hi @charleswilmot 

Unfortunately, I don't think we have already a proper documentation. However, since the `wordlnterface` plugin has been designed and implemented using **Thrift**, it should be fairly easy for the time being to figure out what is the protocol you need to use in order to talk to Gazebo.

All you need is indeed defined in the [`worldinterface_rpc.thrift`][1] file. Each method of the API tells you what are the parameters of the signature you're required to pass via RPC.

For example, if you aim to create a sphere from within C++ code, you have to resort to the [`makeSphere`][2] method with the following snippet:
```cpp
yarp::os::Bottle cmd,rep;
cmd.addDouble(0.1);   // radius [m]
cmd.addDouble(0.1);   // pose.x [m]
cmd.addDouble(0.1);   // pose.y [m]
cmd.addDouble(0.1);   // pose.z [m]
cmd.addDouble(0.1);   // pose.roll [rad]
cmd.addDouble(0.1);   // pose.pitch [rad]
cmd.addDouble(0.1);   // pose.yaw [rad]
cmd.addInt(255);   // color.r
cmd.addInt(255);   // color.g
cmd.addInt(255);   // color.b
// you have also some optional parameters here

gazeboPort.write(cmd,rep);   // talk to Gazebo via RPC

std::string name=rep.get(0).asString();   // name of the object returned by the plugin
```

To know more about our **IDL interface that uses Thrift**, refer to out [documentation][3].

@randaz81 may want to chime in.

[1]: https://github.com/robotology/gazebo-yarp-plugins/blob/master/thrift/worldinterface/worldinterface_rpc.thrift
[2]: https://github.com/robotology/gazebo-yarp-plugins/blob/master/thrift/worldinterface/worldinterface_rpc.thrift#L37
[3]: http://www.yarp.it/tutorials.html

charleswilmot, 2017-03-16:
Hi @pattacini ,

First of all thank you for your quick answer! It's very helpful.

I still need some enlightenment:

- Shouldn't I specify somehow that I want to create a sphere, and not something else? By first adding a String to the Bottle? (`cmd.addString("makeSphere")`)

- Which rpc port should I use? I tried to write on the name server directly but it was not successful. Some issue in the gazebo-yarp-plugin github suggest a `/world` port. I tried to launch gazebo with the option `-slibgazebo_yarp_worldinterface.so`, but it does not seem to create a `/world`. 

Here is the list of my ports:

```
$ ./yarp name list
registration name /icubSim/applyExternalWrench/rpc:i ip 141.2.249.190 port 10081 type tcp
registration name /icubSim/cam/left ip 141.2.249.190 port 10003 type tcp
registration name /icubSim/cam/right ip 141.2.249.190 port 10004 type tcp
registration name /icubSim/head/command:i ip 141.2.249.190 port 10078 type tcp
registration name /icubSim/head/rpc:i ip 141.2.249.190 port 10077 type tcp
registration name /icubSim/head/state:o ip 141.2.249.190 port 10079 type tcp
registration name /icubSim/head/stateExt:o ip 141.2.249.190 port 10080 type tcp
registration name /icubSim/inertial ip 141.2.249.190 port 10002 type tcp
registration name /icubSim/left_arm/analog:o ip 141.2.249.190 port 10006 type tcp
registration name /icubSim/left_arm/analog:o/rpc:i ip 141.2.249.190 port 10005 type tcp
registration name /icubSim/left_arm/command:i ip 141.2.249.190 port 10091 type tcp
registration name /icubSim/left_arm/rpc:i ip 141.2.249.190 port 10090 type tcp
registration name /icubSim/left_arm/state:o ip 141.2.249.190 port 10092 type tcp
registration name /icubSim/left_arm/stateExt:o ip 141.2.249.190 port 10093 type tcp
registration name /icubSim/left_arm_no_hand/command:i ip 141.2.249.190 port 10083 type tcp
registration name /icubSim/left_arm_no_hand/rpc:i ip 141.2.249.190 port 10082 type tcp
registration name /icubSim/left_arm_no_hand/state:o ip 141.2.249.190 port 10084 type tcp
registration name /icubSim/left_arm_no_hand/stateExt:o ip 141.2.249.190 port 10085 type tcp
registration name /icubSim/left_foot/analog:o ip 141.2.249.190 port 10012 type tcp
registration name /icubSim/left_foot/analog:o/rpc:i ip 141.2.249.190 port 10011 type tcp
registration name /icubSim/left_hand/analog:o ip 141.2.249.190 port 10099 type tcp
registration name /icubSim/left_hand/analog:o/rpc:i ip 141.2.249.190 port 10098 type tcp
registration name /icubSim/left_hand/command:i ip 141.2.249.190 port 10046 type tcp
registration name /icubSim/left_hand/rpc:i ip 141.2.249.190 port 10045 type tcp
registration name /icubSim/left_hand/state:o ip 141.2.249.190 port 10047 type tcp
registration name /icubSim/left_hand/stateExt:o ip 141.2.249.190 port 10048 type tcp
registration name /icubSim/left_hand_finger/command:i ip 141.2.249.190 port 10042 type tcp
registration name /icubSim/left_hand_finger/rpc:i ip 141.2.249.190 port 10041 type tcp
registration name /icubSim/left_hand_finger/state:o ip 141.2.249.190 port 10043 type tcp
registration name /icubSim/left_hand_finger/stateExt:o ip 141.2.249.190 port 10044 type tcp
registration name /icubSim/left_hand_index/command:i ip 141.2.249.190 port 10030 type tcp
registration name /icubSim/left_hand_index/rpc:i ip 141.2.249.190 port 10029 type tcp
registration name /icubSim/left_hand_index/state:o ip 141.2.249.190 port 10031 type tcp
registration name /icubSim/left_hand_index/stateExt:o ip 141.2.249.190 port 10032 type tcp
registration name /icubSim/left_hand_middle/command:i ip 141.2.249.190 port 10034 type tcp
registration name /icubSim/left_hand_middle/rpc:i ip 141.2.249.190 port 10033 type tcp
registration name /icubSim/left_hand_middle/state:o ip 141.2.249.190 port 10035 type tcp
registration name /icubSim/left_hand_middle/stateExt:o ip 141.2.249.190 port 10036 type tcp
registration name /icubSim/left_hand_pinky/command:i ip 141.2.249.190 port 10038 type tcp
registration name /icubSim/left_hand_pinky/rpc:i ip 141.2.249.190 port 10037 type tcp
registration name /icubSim/left_hand_pinky/state:o ip 141.2.249.190 port 10039 type tcp
registration name /icubSim/left_hand_pinky/stateExt:o ip 141.2.249.190 port 10040 type tcp
registration name /icubSim/left_hand_thumb/command:i ip 141.2.249.190 port 10026 type tcp
registration name /icubSim/left_hand_thumb/rpc:i ip 141.2.249.190 port 10025 type tcp
registration name /icubSim/left_hand_thumb/state:o ip 141.2.249.190 port 10027 type tcp
registration name /icubSim/left_hand_thumb/stateExt:o ip 141.2.249.190 port 10028 type tcp
registration name /icubSim/left_leg/analog:o ip 141.2.249.190 port 10010 type tcp
registration name /icubSim/left_leg/analog:o/rpc:i ip 141.2.249.190 port 10009 type tcp
registration name /icubSim/left_leg/command:i ip 141.2.249.190 port 10018 type tcp
registration name /icubSim/left_leg/rpc:i ip 141.2.249.190 port 10017 type tcp
registration name /icubSim/left_leg/state:o ip 141.2.249.190 port 10019 type tcp
registration name /icubSim/left_leg/stateExt:o ip 141.2.249.190 port 10020 type tcp
registration name /icubSim/right_arm/analog:o ip 141.2.249.190 port 10008 type tcp
registration name /icubSim/right_arm/analog:o/rpc:i ip 141.2.249.190 port 10007 type tcp
registration name /icubSim/right_arm/command:i ip 141.2.249.190 port 10095 type tcp
registration name /icubSim/right_arm/rpc:i ip 141.2.249.190 port 10094 type tcp
[... 23 lines cut ...]
[... cut ...]
registration name /icubSim/right_leg/stateExt:o ip 141.2.249.190 port 10024 type tcp
registration name /icubSim/torso/command:i ip 141.2.249.190 port 10074 type tcp
registration name /icubSim/torso/rpc:i ip 141.2.249.190 port 10073 type tcp
registration name /icubSim/torso/state:o ip 141.2.249.190 port 10075 type tcp
registration name /icubSim/torso/stateExt:o ip 141.2.249.190 port 10076 type tcp
registration name /root ip 141.2.249.190 port 10000 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message
```

pattacini, 2017-03-16:
@charleswilmot 
- Of course you're right; I've just updated my first post.
- I don't know, I'm not really using Gazebo thus far; @traversaro @randaz81 any idea?

randaz81, 2017-03-21:
Hi @charleswilmot 
you can now check `gazebo-yarp-plugins\tutorial\model\worldInterface` for a working example of world interface plugin

pattacini, 2017-03-27:
Any feedback on this?

pattacini, 2017-03-28:
Closing for the time being; feel free to open it again if required.

charleswilmot, 2017-03-30:
Hello @pattacini ,

I'm sorry, I couldn't answer earlier. My first experiments worked just fine, thank you very much for your precious help!

@randaz81  Thank you for your example, it helped a lot

Best,
Charles

valeriosperati, 2017-07-03:
Please, I need some further clarification about the usage of the world interface plugin. Exactly as @charleswilmot  , I'm looking for a port like "/world", where I can stream commands like "makesphere"  and so on. When using the old simulator the port was already there.  @randaz81 says there is a working example  here

[tutorial/model/worldInterface](https://github.com/robotology/gazebo-yarp-plugins/tree/master/tutorial/model/worldInterface)

but I do not understand the logic:
I can see there are here an urdf file, a config file and  a wordinterface.ini file, not clear to me what I have to do with them.  In the build folder, where I compiled the gazebo-yarp-plugins, I see  a library called libgazebo_yarp_worldinterface.so, exactly in the folder  plugins/worldinterface.

Note that so far  I always run gazebo from command line without any argument; after gazebo is launched, I manually insert the icub model from the gazebo GUI.

I'm wondering if I need to launch gazebo with some command line argument...

System: Linux Debian Jessie, 8.8 , both YARP and gazebo-plugins updated a couple of days ago.
Thanks, Valerio

valeriosperati, 2017-07-04:
Finally I managed to make it work. Thanks a lot @charleswilmot who sent me detailed instructions!
I attach here my working solution:

1) As indicated [here](http://robotology.gitlab.io/docs/gazebo-yarp-plugins/master/embed_plugins.html), I had to append to GAZEBO_MODEL_PATH the path to gazebo-yarp-plugins/tutorial/models. 
where the worldInterface example is located. In this way, the Gazebo GUI shows the World Interface example which, once launched, opens a yarp port "/world_input_port"

2) In order to  enable the port "/world_input_port" without launching the aforementioned example, I prepared a foo.sdf file like this

 ```
<?xml version='1.0'?>
<sdf version='1.4'>
  <world name="my_world">

    <!-- Light -->
    <include>
      <uri>model://sun</uri>
    </include>

    <!-- Ground Plane -->
    <include>
      <uri>model://ground_plane</uri>
    </include>

   <!-- icub  and worldinterface plugin -->
   <model name="iCub_fixed">
   	<include>
      		<uri>model://icub_with_hands</uri>
    	</include>

        <plugin name="world_interface" filename="libgazebo_yarp_worldinterface.so">
           <yarpConfigurationFile>model://worldInterface/conf/worldInterface.ini</yarpConfigurationFile>
        </plugin>

   </model>

</world>
```

Then I launched Gazebo from command line with the file as argument:
`gazebo foo.sdf`

It creates a world with icub inside and the port "world_input_port" available. Then it is easy to create  c++ code that creates and manages objects within gazebo, as @pattacini  says in the example reported above.

closed 2020-12-09 by pattacini