robotology/community #202: PC104 sound

BrutusTT, 2017-03-15:
Hi, 

its the iCubEdinburgh01 again. How can I enable sound on the head?

I tried the yarpdev portaudio according to the wiki page (http://wiki.icub.org/wiki/Grabbing_sound_with_PortAudio):

```icub@icub-head:~$ yarpdev --device portaudio --id 1 --c 2 --read --rate 48000
ALSA lib pcm_dsnoop.c:606:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1029:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_dsnoop.c:606:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1029:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1029:(snd_pcm_dmix_open) unable to open slave
Device number 1
Expression 'parameters->channelCount <= maxChans' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1514
Expression 'ValidateParameters( inputParameters, hostApi, StreamDirection_In )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2818
An error occurred while using the portaudio stream
Error number: -9998
Error message: Invalid number of channels
yarpdev: ***ERROR*** driver <portaudio> was found but could not open
yarpdev: ***ERROR*** driver <grabber> was found but could not open
yarpdev: ***ERROR*** device not available.
Suggestions:
+ Do "yarpdev --list" to see list of supported devices.
+ Or append "--verbose" option to get more information.
```

and also checked the linux related steps:

```
icub@icub-head:~$ sudo su
root@icub-head:/home/icub# modprobe snd-pcm-oss
root@icub-head:/home/icub# modprobe snd-mixer-oss
root@icub-head:/home/icub# cat /dev/sndstat
cat: /dev/sndstat: No such file or directory
root@icub-head:/home/icub# adduser icub audio
The user `icub' is already a member of `audio'.
```

checked that aplay see devices:

```
icub@icub-head:~/software$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA Intel HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: PCH [HDA Intel PCH], device 0: CS4207 Analog [CS4207 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: PCH [HDA Intel PCH], device 1: CS4207 Digital [CS4207 Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: DAC [USB AUDIO    DAC], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

and checked that yarpdev has portaudio support

```
icub@icub-head:~$ yarpdev --list | grep portaudio
Device "portaudio", available on request (found in /usr/local/src/robot/yarp/build/lib/yarp/yarp_portaudio.so library), wrapped by "grabber".
```

So to me it looks like the Ubuntu Installation is missing something. Any idea what that might be?

Thanks in advance.

BrutusTT, 2017-03-15:
After installing pulseaudio and rebooting I could at least get some sound output via the following command which is able to play the provided test wav:

```
icub@icub-head:~$  aplay -D plughw:2,0 test_audio.wav
```

It seems that the default sound card points to the wrong one.

pattacini, 2017-03-21:
Sorry @BrutusTT, it's taking somewhat long to follow this issue because at the moment we have no free resources. Don't despair though, we'll get back to you very soon 😃

randaz81, 2017-03-23:
Hi @BrutusTT,
have you followed the instructions here?
http://wiki.icub.org/wiki/Talking_head

pattacini, 2017-03-27:
Hi @BrutusTT 
any feedback?

pattacini, 2017-03-29:
Closing, feel free to add more on this.

closed 2020-12-15 by pattacini