robotology/community #203: assignment_producer-consumer - Port failed to activate

accateo, 2017-03-18:
Hi, I'm trying to complete the assignment_producer-consumer 
(https://github.com/vvv-school/assignment_producer-consumer).
I have already completed the missing parts of the two .cpp files (producer and consumer).
But when I run 

`./producer`

I have this message

`[INFO]Producer using sync mode 
yarp: Port /producer/out failed to activate at tcp://192.168.1.4:10002/ (address conflict)
[ERROR]cannot open the output port`

Why? I have tried to fix using this method:
http://wiki.icub.org/wiki/Conflicts
but nothing.
Thanks for patience!

pattacini, 2017-03-18:
Hi @accateo 

Please refer to a complete handling of the _address conflitct_ problem as per #181

pattacini, 2017-03-18:
Personally, I'd first verify whether your code is trying to open the same port twice and then check if a system restart will sort out the conflict. Then, I'll read carefully the instructions provided in #181.

accateo, 2017-03-19:
With a reboot of the entire system, producer and consumer works correctly.
When I try to run:

`connect /producer/out /consumer/in tcp`

I have:

`ERROR: You must specify the destination port correctly.
usage: connect [-dnhst45] [-p local-port][-R resolve] [-w timeout] 
          [-H proxy-server[:port]] [-S [user@]socks-server[:port]] 
          [-T proxy-server[:port]]
          [-c telnet-proxy-command]
          host port
`

It should not be "yarp connect..."?
Thanks

pattacini, 2017-03-19:
Of course it's `yarp connect`.
Typos fixed in https://github.com/vvv-school/assignment_producer-consumer/commit/21cf4b2c0f0f2d128db540dbe7402eb3a0676812.

Thanks @accateo for pointing them out.

accateo, 2017-03-19:
You're welcome

closed 2020-12-08 by pattacini