robotology/community #204: Precautions before using codyco-modules on real iCub

asutcl, 2017-03-22:
Hi,

We currently have the code running the codyco modules in simulation. We have a harness for the iCub and we are planning on testing the torqueBalancing module on the real robot. 

Are there any additional precautions that should be taken before running the codyco-modules on the iCub?

Thank you,
Andrew

pattacini, 2017-03-22:
I think this request goes to @robotology/codyco-developers. 
Anyone of the team showing up with an appropriate answer?

pattacini, 2017-03-22:
Btw, @asutcl you should have received the invitation by email to join our community so that you'll be able to categorize issues with labels.

asutcl, 2017-03-22:
@pattacini, thank you! Sorry this is the second time you invited me. I hadn't properly followed the link the first time before the invitation expired.

traversaro, 2017-03-22:
Hi @asutcl , the `torqueBalancing` C++ module has not been tested on the real robot for a long time, unfortunately. 

However, last year in Darmstadt we tried to document some of the preliminary steps. The document is available at https://github.com/robotology/codyco-superbuild/pull/120/files and still needs to be cleaned up, but I think it is a good  general overview of what you need to check before trying stuff on the real robot. 

A preliminary check that I would reccomend is to make sure that your iCub's URDF model has been updated recently. Are you working on `iCubDarmstadt01`?

pattacini, 2017-03-22:
@traversaro they're users of **iCubLausanne01/02**.

asutcl, 2017-03-23:
Hi @traversaro,

Thank you for the info. As @patticini pointed out we are working with iCubLausanne02. We are interested in making the robot stand. Are we correct in assuming that the torqueBalancing module is the easiest way to do so?

francesco-romano, 2017-03-27:
Hi @asutcl 

TorqueBalancing (both the (unfortunately old) C++ implementation and the newer Simulink one) leverages the torque capabilities of iCub.
Unfortunately to obtain a nice and reliable torque control some motor parameters must be identified beforehand.

We are currently working on creating a Matlab script to perform the identification on a robot similar to iCubLausanne. When we will have progress on this activity you may want to try the identification procedure on your robot.

We will keep you posted!

cc @DanielePucci @traversaro @S-Dafarra

asutcl, 2017-03-28:
Hi @francesco-romano,

Thank you for the update. Sounds great! We will wait for the identification script.

serena-ivaldi, 2017-03-28:
Please keep us in the loop of this activity because we have the same issue here, we want to do the identification then having the torque balancing... 
@InriaBrice @dogoepp @olivierrochel-inria @jbmouret

pattacini, 2017-03-31:
I would close this issue and let @francesco-romano keep you better posted with the issue tracking system of the dedicated repository.

closed 2020-12-08 by pattacini