robotology/community #205: Use of ESD CAN-USB/2 on Windows with YARP

traversaro, 2017-03-27:
Hi everyone, I have a ESD CAN-USB/2 (the blue one), I am trying to get it to compile with Windows, as if I remember correctly, such device should be working fine in Windows.
In the (I supposed outdated) documentation in http://wiki.icub.org/wiki/Esd_on_windows , it is written that the `ecan` YARP device present in `icub-main` supports only the ESD CAN-USB-MINI . 
Assuming that information was wrong, I downloaded the CAN-USB/2 SDK from https://esd.eu/en/products/can-usb2 and I ensure that CMake is finding the SDK. 
However when compiling the `ecan` device in `icub-main` I get the following compilation errors:
~~~
Build FAILED.

"C:\Code\Production\icub-main\build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Code\Production\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj" (default target) (20) ->
(ClCompile target) ->
  C:\Code\Production\icub-main\src\libraries\icubmod\esdCan\EsdCan.cpp(59): error C2664: 'NTCAN_RESULT canGetBaudrate(NTCAN_HANDLE,uint32_t *)' : cannot convert argument 2 from 'DWORD *' to 'uint32_t *' [C:\Code\Production\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj]
  C:\Code\Production\icub-main\src\libraries\icubmod\esdCan\EsdCan.cpp(100): error C2664: 'NTCAN_RESULT canRead(NTCAN_HANDLE,CMSG *,int32_t *,OVERLAPPED *)' : cannot convert argument 3 from 'long *' to 'int32_t *' [C:\Code\Production\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj]
  C:\Code\Production\icub-main\src\libraries\icubmod\esdCan\EsdCan.cpp(102): error C2664: 'NTCAN_RESULT canTake(NTCAN_HANDLE,CMSG *,int32_t *)' : cannot convert argument 3 from 'long *' to 'int32_t *' [C:\Code\Production\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj]
  C:\Code\Production\icub-main\src\libraries\icubmod\esdCan\EsdCan.cpp(129): error C2664: 'NTCAN_RESULT canWrite(NTCAN_HANDLE,CMSG *,int32_t *,OVERLAPPED *)' : cannot convert argument 3 from 'long *' to 'int32_t *' [C:\Code\Production\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj]

    0 Warning(s)
    4 Error(s)
~~~

traversaro, 2017-03-27:
I received various feedback and apparently this is indeed a bug in the `master` branch of icub-main the `ecan` device driver, moving the issue to `icub-main` to discuss about it.
In a nutshell, moving to the `devel` branch (or cherry-picking this commit: https://github.com/robotology/icub-main/commit/72261273f4e60eb65230a34ec8f383cfb9c04440 ) should solve the issue.,

pattacini, 2017-03-27:
Issue moved to [robotology/icub-main #434](https://github.com/robotology/icub-main/issues/434) via [**ZenHub**](https://www.zenhub.com/)

closed 2020-12-15 by pattacini