robotology/community #206: yarpmanager and stdio tag

valeriosperati, 2017-03-27:
Hi, I wrote a simple xml file read by yarpmanager. The file successfully launch the simulator on the node "nodo_valerio" .

```
<application>
<name>Valerio example</name>
<dependencies>
      <port>/nodo_valerio</port>
</dependencies>

    <module>
        <name>iCub_SIM</name>
        <parameters></parameters>
        <node>nodo_valerio</node>
        <stdio>/nodo_valerio</stdio>
        <tag>simulator</tag>
    </module>
</application>
```
Node was created launching 
`yarprun --server /nodo_valerio`

Now, I would like to see the console output of the simulator on a console of the same node.
As you can see, I added <stdio> parameter with the node name in xml file. Where should the console output appear? I would expect it where I launched yarprun, but I'm clearly wrong.
Thanks, valerio

pattacini, 2017-03-27:
Hi @valeriosperati 

I honestly don't know whether the `stdio` option is still functional. In the past, I happen to use it only with the console as the output node; I've never tested it in different configurations.

Unless you have a particular goal in mind, the latest recommended way to log modules' printouts is through the `yarplogger` utility. You have to meet two requirements to this end:

1. Launch the `yarprun` in log mode: `yarprun --server /node-name --log`
2. Within your modules rely on `yWarning`, `yDebug`, `yInfo`... services in order to redirect the output.

Related issue: #170 

cc @apaikan

valeriosperati, 2017-03-27:
Ok, I'll try. Is there some documentation about yarplogger? The page
[www.yarp.it/yarplogger.html](url)
is empty. Is there some example on how to redirect the output?
Thanks!

pattacini, 2017-03-27:
Unfortunately, we don't have a proper documentation yet (see https://github.com/robotology/yarp/issues/870). 

However, from the code, it's quite simple to redirect the output. Just do:
```cpp
#include <yarp/os/LogStream.h>
using namespace yarp::os;
// ...
yInfo()<<"This is a [INFO] message";
yDebug()<<"This is a [DEBUG] message";
yWarning()<<"This is a [WARNING] message";
// ...
```

From the xml you don't have to do anything special; only make sure that the `yarprun` instances are launched with the `--log` option.

Before launching your application, start the `yarplogger` tool and press on the **play** button.

valeriosperati, 2017-03-27:
Ok, thanks for the code, but I guess it is ok only for custom code, where I can insert these lines. How can I do with the iCub_SIM?

pattacini, 2017-03-27:
`iCub_SIM` should be already using the new logging functionalities. Anyway, don't worry, `yarplogger` is smart enough to catch also those printouts that are not nicely formatted (e.g. `cout`, `cerr`, `printf`...), although keep in mind that we discourage their usage.

valeriosperati, 2017-03-28:
mmm... I'm missing something. I did run this test: 

1) launched yarprun
`yarprun --server /nodo_valerio --log`

2) launched yarplogger
`yarplogger`

3)  launched yarpmanager reading my custom xml file, (it will start icub_SIM and yarpview, when requested)

4) start the logger, hitting "Play" on the logger interface

5)  run the icub_SIM on yarpmanager

6) finally, I hit "Save Log Session" on logger interface 

The save log file simply display
`10`
No info about the simulator.

In a second test, if I run the icub_SIM with yarpview module and save the log, I have some info about yarpview:

```
1
1
150.146.65.56
/log/nodo_valerio/yarpview/11913
/nodo_valerio
log
yarpview
11913
0
0
0
0
0
0
4
4
0
0.000001
0
<#STRING_START#>/icub_left_modified
<#STRING_END#>
1
0.000582
0
<#STRING_START#>yarp: Port /icub_left_modified active at tcp://150.146.65.56:10055/
<#STRING_END#>
2
5.276665
0
<#STRING_START#>yarp: Receiving input from /icubSim/cam/left to /icub_left_modified using udp
<#STRING_END#>
3
5.407396
0
<#STRING_START#>Surface STARTED! Dimensions: 320x240 -- PixelFormat: 3
<#STRING_END#>
```

In both cases, no info about the simulator. I think that logger takes only some custom messages (as you suggested), and not the console output. Is it correct?

pattacini, 2017-03-28:
Well, I ran a quick test on my side with the following simple application:
```xml
<application>
    <name>Test</name>
    <dependencies>
    </dependencies>
    <module>
        <name>iCub_SIM</name>
        <dependencies>
            <port timeout="20">/yarplogger</port>
        </dependencies>
        <node>pattacini</node>
    </module>
</application>
```
obtaining the attached [log_pattacini_iCub_SIM_1752.txt](https://github.com/robotology/QA/files/875381/log_pattacini_iCub_SIM_1752.txt), which does contain all the required info.

I'm using `yarp/devel` against `icub-main/devel` on a Windows machine.

In particular, what I did was just right-clicking on the log (top-left panel) to then select the option to save on file.

valeriosperati, 2017-03-28:
Thanks! I managed to make it work. What was missing in my xml file was the line
```
<dependencies>
<port timeout="20">/yarplogger</port>
</dependencies>
```
in the module description icub_SIM.
Test run on Debian 8.7 Jessie.

valeriosperati, 2017-03-28:
Closing the issue :)

closed 2020-12-08 by pattacini