robotology/community #210: Bypass the PID controler

charleswilmot, 2017-04-04:
Dear all,

I would like to control a cube floating in the vacuum in Gazebo by hard-setting the position and the velocity (no real motor). For now I'm moving the cube using PID controllers. The cube is connected to the `world link` of my model by 3 prismatic joints.

Would the new method [`disablePID`](http://www.yarp.it/classyarp_1_1dev_1_1IPidControl.html#ad872945ab25bd4976898c2ff674dd58c) do what I want? _"Disable the pid computation for a joint"_

Should I rather try to use the [`PositionDirect`](http://www.yarp.it/classyarp_1_1dev_1_1IPositionDirect.html#details) interface? _"the aim is to reach low level control in firmware bypassing the trajetory generator"_

Is there a more appropriate method for doing that?

Best

pattacini, 2017-04-04:
Hi @charleswilmot,

What you're trying to achieve is to fully bypass the motor, by imposing the state of the cube straightaway. Thereby, you don't have to rely on any of our interfaces, but rather you should be fiddling within Gazebo directly, I suppose.

pattacini, 2017-04-04:
I believe you should go in the direction of using the world interface instead (see #201).

charleswilmot, 2017-04-18:
Hello @pattacini ,

Sorry for the late answer, I have been busy working on the real robot last weeks. I'm still very interested in bypassing the motor in the simulator. Indeed, my first intuition was to use the world interface plugin. Thanks to your help on the forum here, I got it working, unfortunately one functionality is missing for my purpose. I need to control an object's speed is a similar way as the function `setPose` of the `worldinterface` plugin controls the position.
I would like to implement it within the `worldinterface` plugin, but I would need some more advises.
I'm not familiar with the `thrift` technology. What would be the basic procedure to add a new function to the plugin?

Thank you in advance,
charleswilmot

pattacini, 2017-04-19:
Hi @charleswilmot 

Did you have the chance to peruse our [documentation and tutorials][1] on Thrift?

[1]: http://www.yarp.it/tutorials.html

charleswilmot, 2017-04-19:
Hello @pattacini , sorry I didn't found that in the doc before. I'll read it next week.

Thank you once more

closed 2020-12-08 by pattacini