robotology/community #214: Dragonfly Parameters documentation not available

pedrombmachado, 2017-04-20:
Dear all,
   For some reason the link http://eris.liralab.it/iCub/main/dox/html/group__icub__dragonfly2.html is not working. The link is suggested by the wiki http://wiki.icub.org/wiki/Dragonfly_Parameters.

pattacini, 2017-04-20:
Hi @pedrombmachado 

Some time ago we migrated our infrastructure to http://wiki.icub.org.

Therefore, try this out: http://wiki.icub.org/iCub/main/dox/html/group__icub__dragonfly2.html.

I've updated the wiki accordingly.

pedrombmachado, 2017-04-20:
Hi @pattacini ,
  It works. Many thanks.

frank-foerster, 2019-03-05:
Hi (@pattacini ?), with a bit less than two years delay: the above website is offline again.
Could someone with the powers to do so, bring it back online again? 
Many thanks.

traversaro, 2019-03-05:
Sorry for the trouble @frank-foerster , apparently the docs have changed the URL again, the documentation you are searching for should be available at http://www.icub.org/software_documentation/group__icub__dragonfly2.html .

frank-foerster, 2019-03-05:
Great, thanks! By the looks of it the cameras can't do `native' black & white though, so I'll have to pump colour images through the network :(. 

Thanks again.

closed 2020-12-15 by pattacini