robotology/community #216: yarpmanager make it roll assignment

accateo, 2017-04-25:
Hi,
I resolved the make it roll assignment:
https://github.com/vvv-school/assignment_make-it-roll
and I want to test my solution using yarpmanager. So I have executed yarpmanager  and I imported the two applications in app folder.
The "assignment_make-it-roll-app" works fine, but "assignment_make-it-roll-system" not.
When I run the second, starts only yarplogger, while the other modules go in timeout waiting a resources.
What resources do I need?
Thanks

pattacini, 2017-04-25:
@Arya07 @Iaxama @gmmarconi @massimoregoli @towardthesea @giuliavezzani 
anyone willing to help?

Iaxama, 2017-04-26:
While the yarplogger does not have any dependency, all of the other modules depend on the iCubSim to be up and running to start. Try to run iCubSim from a terminal and check that you can run it properly, then try again.

accateo, 2017-04-29:
Ok, now it works. I have also sent rpc command to the robot, in this way the robot look down.
Now the robot should search the blue ball or I have to move the head to the right?

Iaxama, 2017-05-02:
You can use the getCog method provided in the main.cpp to get the center of the ball in the two images. Then you have to write the code to fixate at that point (Hint: use the IGazeControl to do that)

accateo, 2017-05-06:
No ok, I already use getCog and IGazeControl in the code, but when I run it the robot looks forward. 
I run yarpmanager with the two modules, yarpserver and the compiled main.cpp code

Iaxama, 2017-05-06:
For the getCog method to work the ball should be in the visual range of the cameras. The robot has to look down towards the table first, then get the ball position in the two images with the getCog method, then you need to retrieve the ball position using the iGazeControl.triangulate3dPoint method, then ask the robot to approach the ball position with the hand and push it.

accateo, 2017-05-13:
Ok, thanks!

pattacini, 2017-05-20:
Closing for now; feel free to open it again if needed.

closed 2020-12-15 by pattacini