robotology/community #217: Texture on the ICub Simulator

LakshithaPW, 2017-04-29:
Hello,

I have a query regarding the texture on the ICub simulator  (Specifically the Bluetexture on the cover).
When blue texture is applied, it will be applied on the robot inside-out which is a strange thing.
I was wondering whether there is some solution without reinstalling the iCub Simulator.
Please let me know if you have some idea about the reason for this.

Thanks
lakshitha

vtikha, 2017-05-22:
@LakshithaPW No, the blue texture is also applied on the inside of the cover. Maybe I did not fully understand but I do not see it as a strange thing eg: this also is the same thing on the real robot.

LakshithaPW, 2017-05-25:
@vtikha Thanks a lot, I tried to reinstall the simulator and it seems to be working correctly now.

closed 2020-12-09 by pattacini