robotology/community #218: Cannot make connection to 'xmlrpc

pedrombmachado, 2017-05-02:
Dear All,
 
  I am trying to configure YARP to work ROS but I am getting the error:
Cannot connect to ROS subscriber http://cncr-1:41205/
Cannot make connection to 'xmlrpc://cncr-1:41205'

NOTE: Open issue on YARP github https://github.com/robotology/yarp/issues/1190

[image]

I have compiled yarp from the sources and I followed the instructions described [here](http://wiki.icub.org/wiki/YARP_ROS_Interoperation_Past_and_Future_But_Not_Now) and [here](http://www.yarp.it/yarp_with_ros_nameservers.html). 

Please find below CMakeCache.txt used to compile yarp:
[CMakeCache.txt](https://github.com/robotology/yarp/files/967562/CMakeCache.txt)

```
pedro@cncr-1:~$ yarp version
YARP version 2.3.68+230-20170426.1+git4c562a6+dirty
```

ros_comm version 1.12.7

```
Environmental vars on .bashrc:
export YARP_DIR="/usr/local"
export YARP_DATA_DIRS="$YARP_DIR/share/yarp"
export PATH="$PATH:$YARP_DIR/bin"
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:YARP_DIR/lib"
```

```
pedro@cncr-1:~$ locate xmlrpc.ini
/usr/local/share/yarp/plugins/xmlrpc.ini
/usr/share/yarp/plugins/xmlrpc.ini
```

Linux cncr-1 4.8.0-49-generic #52~16.04.1-Ubuntu SMP Thu Apr 20 10:55:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

pattacini, 2017-05-02:
Hi @pedrombmachado 
Having duplicated issues around is bad for maintenance. Keep in mind this for the future, please 😃

pattacini, 2017-05-02:
In particular, https://github.com/robotology/yarp/issues is a good place for it.

pedrombmachado, 2017-05-05:
Hi @pattacini ,
  Sorry about that. The problem is that no one is answering my question on the Yarp issues. Can you please find someone to take care of that issue?

pattacini, 2017-05-05:
Hi @pedrombmachado 

Consider that we are not a company and we have limited resources for doing research and providing support at the same time; as a consequence, it might happen that we are not always super responsive 😄 and this has a clear impact on the user workflow.

Nonetheless, to alleviate these undesired effects, we've been continuously improving our internal processes and, of course, we can also get significant benefits from the help of the community (thanks @plinioMoreno in this particular case).

pedrombmachado, 2017-05-05:
Hi @pattacini ,
   I know we are all in the same boat. :p That is why I asked your help.
Many thanks for this

closed 2020-12-08 by pattacini