robotology/community #220: Yarp/iCub dependencies - Any plans for MSVC14

kt10aan, 2017-05-09:
Is there any plan for having installer, or just individual links for the YARP/iCub dependencies compiled with MSVC14 (Visual Studio 2015)?
Especially the iCub dependencies, as they are much harder to compile on Windows (GSL, Ipopt)

I found the ode library compiled for msvc14 on the wiki, so I assume someone must be using MSVC14 for his YARP/iCub needs :)

pattacini, 2017-05-09:
Hi @kt10aan,

Yep, we're working on it and our goal is to provide new YARP/iCub releases compatible with Visual Studio 2015 asap.

drdanz, 2017-05-10:
@kt10aan Meanwhile, if you don't need `ipopt`(unfortunately it does not work yet), you can use [`vcpkg`](https://github.com/Microsoft/vcpkg), all the other dependencies are there.

kt10aan, 2017-05-12:
@drdanz, thanks for the tip, the `vcpkg` project looks very promising :)

traversaro, 2017-05-12:
Actually we are also missing `sdl1` w.r.t. to the dependencies that we ship with the installer. 
All other dependencies should be there, i.e. : 
~~~
./vcpkg install --triplet <selected_triplet> ace eigen3 qt5 
~~~
for YARP dependencies and 
~~~
./vcpkg install --triplet <selected_triplet> ace eigen3 qt5 ode gsl opencv freeglut
~~~
for YARP & iCub dependencies.
Please note that this will compile all the dependencies from source code, so compiling bit packages such as Qt5 could take several hours on an average computer.

closed 2020-12-15 by pattacini