robotology/community #221: Q&A - FW update of the eLoader and eUpdater in ETH robots: is it to do?

marcoaccame, 2017-05-10:
Dear ETH-iCub-ers,

next release of iCub will introduce a new tool for FW update: the program *FirmwareUpdater*. 

We have designed this new tool to be backwards compatible and as such it recognizes and programs every board mounted on the robots around, even if the FW of the boards is not aligned to master.

However, to offer some useful new features in the ETH robots, the *FirmwareUpdater* requires to find ETH boards (e.g., EMS and MC4PLUS) which have a recent version of the eLoader and eUpdater, and in particular the versions which have been in master since December 2016.

With this post I offer information about  FW upgrade of the ETH robots to the latest eLoader and eUpdater. 

What are features that they enable? How long or difficult is the FW upgrade? Is it worth doing the upgrade? How and when to contact me/others to help upgrading? We shall see that in the following.

**What are the new features enabled by the latest eLoader / eUpdater?**

Here are the new features: 
* the ETH board always tells what kind of board it is and hence it is impossible to confuse an EMS with a MC4PLUS and hence program in them the same FW.
* The ETH board now offers a R/W Info field in internal EEPROM.
Board 10.0.1.1 has these features, board 10.0.1.2 has old versions of eLoader / eUpdater.

[image]
**Figure 1** - FirmwareUpdater in GUI mode. The type of board and the Info field is displayed only for recent versions of eLoader / eUpdater, e.g., as for board 10.0.1.1

[image]
**Figure 2** - FirmwareUpdater in GUI mode. The board 10.0.1.2 has old versions of eLoader / eUpdater.

**How long or difficult is the FW upgrade?**

For each board separately we need to:
1. Program the new eLoader, restart the board, check its IP address. If 10.0.1.99 then change its IP address back to its original.
2. Load and run a special application (e.g., the emsApplPROGupdater.hex for EMS) and use it to program the new eUpdater. Verify that the new eUpdater runs correctly.
3. Reprogram back the normal application (e.g., ems.hex for the EMS).

Here are the only two risks which I can think of:

* If you don’t do the update one by one, then you may end up with all boards having IP address equal to 10.0.1.99. To recover from that is possible, but longer. 
* If your robot is not aligned to master, then before any operation you must write down the current application FW version and possibly also the SHA of icub-firmware-build from where those FW came (or simply copy the entire tree somewhere else).

In short: that is not difficult or long or too risky but **I strongly advise that you do it together with a person of icub-support which will take care of critical passages**. 

**Is it worth doing the upgrade?**

Well, it is up to your attitude, to critical activities you have in your lab. 

Surely, I strongly advise to perform this operation only after a full alignment to master.

**How and when to contact me to help upgrading?**

If you decide that you want to perform the full FW update to eLoader and eUpdater, you can contact me in several ways: 
1. By direct email request to marco.accame@iit.it
2. By opening an issue on icub-support with me in cc with topic: `FW update of eLoader and eUpdater: iCubYourNameOfRobot`, or similar. One issue per robot! So the we can open and close only for those of you interested in doing the operation.

I will open two separate issues for iCubGenova02 and iCubGenova04, even if they already have the latest FW. They will be used as an example and for you as a guide to look at details of the activities.

There is no need to perform the update now or before the next release of iCub (which will be maybe in mid 2017), hence you can contact me or icub-support also later in the full 2017.

marcoaccame, 2017-05-10:
I have already opened two issue for our robots in Genova: https://github.com/robotology/icub-support/issues/455 and https://github.com/robotology/icub-support/issues/456.

Here are some persons who I recall were interested in the issue of FW update of eLoader and eUpdater:
* issue #151: @hu-yue (robot `iCubHeidelberg01`)
* issue #134: @serena-ivaldi , @paraschos (robot `iCubNancy01`).

And here is the list of all the other ETH robots which may need the update and for which I don't have a name to alert directly.
* `iCubDarmstadt01`
* `iCubEdimburgh01`
* `iCubHamburg01`
* `iCubLausanne02`
* `iCubNottingham01`
* `iCubSheffield01`
* `iCubSingapore01`
* `iCubTwente01`

traversaro, 2017-05-10:
@paraschos is actually working in Darmstadt .

marcoaccame, 2017-05-11:
corrected. thanks.

marcoaccame, 2017-05-12:
Hi all, 
here is how to check yourself what FW versions are running on your robot. If you are in doubt, you can follow these operations and then send me the log. I will tell you if you need or not the full FW update.

1.	Switch motors OFF, then ON again, waits some seconds (> 3 is fine).
2.	Launch ethLoader.
3.	Press Discover and wait some seconds (>3 is fine). Press Discover again and wait some seconds. At this time all the ETH boards will be displayed on the GUI and they will be running the eUpdater.
4.	Select all the boards by pressing the Select All button, and press button More Details.
5.	The result will be that every board sends its FW information to eth ethLoader which prints it on the right part of the GUI.
6.	Select and copy all the text on the right part of the GUI and send it to me via email (marco.accame@iit.it).
7.	From this log I will be able to understand which firmware is running on every board.

closed 2020-12-12 by pattacini